cedricp/ddt4all #298: Window Closing on Clio IV ph2

tergin76, 2019-11-23:
Hi;

I enabled the setting under T4_VS_BCM_DDT2000_SW16_5->Windows Lift->AUTO_WINDOW_CLOSING false to true. But even I try to double or triple press on locking button, opened front and rear windows are not closing. Do I have to change any other settings under anywhere on another menu? 

[image]

Any idea?

Thanks

tergin76, 2019-11-25:
As I research, there should be an external module for "Auto Window Closing" feature installed seperatly for this option to work. Like "Enable Rear Parking Camera" option. If there is no camera, what will it be to enable. So, please corret me, if I'm wrong.

tergin76, 2019-11-25:
That is what I mean;

WINSGO Auto Car Power Window Closer Automatically for Renault Koleos 2016+/Kadjar/Subaru XV 2
[https://www.aliexpress.com/item/32871739910.html](url)

michalzahor, 2020-02-01:
I believe not, this product just get the signal that you lock the car and auto close windows. The T4_VS_BCM_DDT2000_SW16_5->Windows Lift->AUTO_WINDOW_CLOSING should set this behaviour in existing ECU in the car. For me, in Koleos 2018, it is also not working.

m0xz0r, 2020-02-08:
From what I've read somewhere, the AUTO_WINDOW_CLOSING parameter only works when an official Renault alarm system (e.g. part# 8201656123) has been installed.

michalzahor, 2020-02-08:
I have original Renault alarm (don't know part number) - but auto windows closing not working

Michal Zahor
mzahor@dtech.sk

Doowayp, 2020-03-28:
D'après des informations que j'ai pu lire sur des forums, il faut avoir les boutons électrique à l'avant et à l'arrière avec les 2 crans.
Corrigez moi si c'est faux..
Mais j'ai activer cette option, mais cela ne fonctionne pas non plus pour moi.. déçu.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif