cedricp/ddt4all #299: Ecu-tweaker 1.81 crash

severusnt, 2019-11-24:
crashed under Android 9.0 immediately after aplication start even with elm237 removed

severusnt, 2019-11-24:
From CatLog
11-24 21:40:02.240 W/m       (27561): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)

tyronecarrion, 2020-01-29:
Same here, but in my case it crashes only when selecting USB

cedricp, 2020-01-29:
@severusnt Could you catlog with this debug version : 
https://drive.google.com/open?id=1x2J8-ckAoPugw3Q01-rcaigZ7eQdHZTz

tyronecarrion, 2020-01-29:
Ok, Will do

Atenciosamente,
Tyrone Carrion

@severusnt Could you catlog with this debug version :
https://drive.google.com/open?id=1x2J8-ckAoPugw3Q01-rcaigZ7eQdHZTz

trplsn, 2022-08-13:
@cedricp ciao ho installato app ECU-Tweaker su telefono android, riesco a collegarmi a ODB2 wifi e riesco a leggere tutti parametri ma non riesco a modificarli perché mi dice ERROR NO DATA. Sai dirmi il perché? Grazie

AndyIreland, 2022-08-13:
> hello I have installed ECU-Tweaker app on android phone, I can connect to ODB2 wifi and I can read all parameters but I can't modify them because it tells me ERROR NO DATA. Can you tell me why? Thank you

what car have you got and what year?, it may have a gateway fitted to the CAN system whereby you can read parameters but to modify parameters you have to buy a token to unlock the gateway

trplsn, 2022-08-13:
Ho una Renault Scenic IV del 2017, ho fatto il cambio del display con
quello da 8,7 pollici ma non riesco a modificare i parametri

trplsn, 2022-08-13:
Come posso acquistare questo token? Mi potete aiutare per favore?

AndyIreland, 2022-08-13:
https://asos.renault.com/#panel3v 

if you need to unlock gateway - I think Scenic IV Ph2 is Gateway locked ... but dont take my word for it

trplsn, 2022-08-13:
Assurdo, ci sono un sacco di tutorial che spiegano come effettuare le modifiche tramite ddt4all con Renault Scenic IV ma nessuno ha mai parlato di sbloccare il gateway. Non vorrei che fosse un problema legato al apk ECU-Tweaker.

KarelSvo, 2022-08-14:
2017 is not phase 2 ;-) Without log data and Elm test results, your problem cannot be defined and cannot be solved

trplsn, 2022-08-14:
Ok proverò ad inserire i dati di registro

Sartascn, 2023-03-11:
> 

stesso problema! Mi sapreste dire come avete risolto? Grazie

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif