cedricp/ddt4all #300: Megane 3 option to enable folding mirrors

charliehoward, 2019-11-25:
I'm going to be installing folding mirrors but cannot find any options that look like folding mirrors, does anyone know where it is?

tergin76, 2019-11-26:
Hi;

If your car has a folding/unfolding capability, it should be under **UPC-EMM** section as **Auto_Folding**.

BarektE, 2019-11-26:
e-mail yaz özelden tarif edeyim

charliehoward, 2019-11-26:
Email what?
I can't find any options in UPC that look like folding but I did find a "folding" option in DDCM, would that be it?

BarektE, 2019-11-27:
UPC do not already have the man crate you write the e-mail address where the active show where we said, but did not care, many tinnnn no longer describe the e-mail göstersende.

BarektE, 2019-11-27:
UPC does not already exist. I don't think you need it.

charliehoward, 2019-11-28:
Email me and we can discuss?

charliehoward, 2019-11-28:
For reference I'm using the X95 ECU

BarektE, 2019-11-28:
Just in case you don't need it.
[image]

BarektE, 2019-11-28:
connection diagram

[image]

charliehoward, 2019-11-28:
I've found that in DDCM, is that right? It also says ECU requires reset after EEPROM writing paramater at the bottom, would this wipe a ECU map?

BarektE, 2019-11-28:
Sonradan takıyorsan buda lazım olacaktır

[image]

BarektE, 2019-11-28:
hepsi birden programlanması gerekiyor tek programlar san o bölümdeki her şey iptal olacaktır. Tek denemeyin.

charliehoward, 2019-12-04:
So I tried programming it tonight and when I scanned for ECU's it found [PDCM] Pass DCM E95 SLOT v4.0 but there is no config, there is only read.
[A photo of the issue](https://i.imgur.com/cAkZEkP.jpg)
I also tried ones it couldn't find and it did change the values but they still do not fold.

kadirayhantr, 2020-01-24:
> UPC zaten mevcut değil. İhtiyacın olduğunu sanmıyorum.
> İhtiyacınız olursa bana posta ile yazarsınız.
> ama artık yazmaya yardımcı olmayacak.

Hi, kadirayhantr@gmail.com  is my mail adresse. Can you tell me how can I programme my  UPC ?

kadirayhantr, 2020-01-24:
> hepsi birden programlanması gerekiyor tek programlar san o bölümdeki her şey iptal olacaktır. Tek denemeyin.

536 969 55 27 bu işlerden anlayan birine benziyorsunuz. Watsap tan görüşebilir miyiz? Teşekkürler

johnroberts2k, 2020-08-08:
Hey @charliehoward just came across this post. I am trying to do exactly that. My Megane 3 didn't come with folding mirrors, but I've purchased a pair of folding mirrors which seem to have an extra pin for the mirror motor. The wire coming from the driversdoor multiplex computer seems to have a wire for the motor pin. I also have the button on the window switch for folding and unfolding mirror. It doesn't work, so I am thinking maybe it is turned off somewhere in the ECU. The wiring is there but the car seems to be ignoring it.

Did you get anywhere with yours?

charliehoward, 2020-08-08:
Yeah it needs to be coded in, follow my guide here - https://rsmegane.com/threads/retrofit-folding-mirrors.17826/

johnroberts2k, 2020-08-08:
@charliehoward awesome man that is great. you've made my day :)

johnroberts2k, 2020-08-10:
@charliehoward just wondering at the bottom of the DCM write section, what does it mean when it says: Warning: ECU reset is needed after each EEPROM writing parameter set to be apply by the SW?

From what I can see it sends all the parameters as a single send command right? Some sections when you work on configs they let you send each parameter and save it. The DCM write seems like you set everything and then press send right?

BarektE, 2020-08-11:
> Hey @charliehoward az önce bu yazıya rastladı. Ben de tam olarak bunu yapmaya çalışıyorum. Megane 3'ümde katlanır aynalar yoktu, ancak ayna motoru için fazladan bir pimi varmış gibi görünen bir çift katlanır ayna satın aldım. Sürücü kapısı multipleks bilgisayardan gelen telde motor pimi için bir tel var gibi görünüyor. Aynayı katlamak ve açmak için pencere anahtarında da düğme var. Çalışmıyor, bu yüzden ECU'da bir yerlerde kapatılmış olabilir diye düşünüyorum. Kablolar orada ama araba onu görmezden geliyor gibi görünüyor.
> 
> Seninkiyle bir yere vardın mı?
> 
> Teşekkürler

Merhaba e posta yaz tarif edeyim

closed 2020-08-14 by charliehoward

EmreMegane, 2023-07-14:
> e-mail yaz özelden tarif edeyim

Merhabalar benim aracımda da aynı sorun var ddt4all ile OBD II cihazını bağladım sonra aynalarımda arıza çıktı bana yardımcı olurmusunuz