cedricp/ddt4all #303: Virgin mode megane iv rlink 2

maks88888, 2019-12-08:
Megane iv 2016
Virgin mode rlink 2 
hello
I selected the virgin mode (developer mode).
Please help to exit this virgin mode.
Thanks

maks88888, 2019-12-08:
display 7 '
thank you very much for help

shrlnm, 2019-12-09:
> Please help to exit this virgin mode.

e-mail?

maks88888, 2019-12-09:
Maks888@o2.pl

BillyJac, 2019-12-20:
Hi @shrlnm ,
Firstly, thank you so much for the amazing work with all your help (Read ALOT of your comments.lol)
I have Megane IV 2017, 
Pressed HIGH in Developer mode, then attempted manual firmware update 2.300 -> 3.3 
Unit started update and restarted - it never came on again. Screen dead
I have ELM327 (Wifi), Win10

Again, thank you for your help
billy.jacobie@gmial.com

engindalmis, 2019-12-21:
engindalmis26@hotmail.com

shrlnm, 2019-12-22:
> [engindalmis26@hotmail.com](mailto:engindalmis26@hotmail.com)

Could you please explain your issue

engindalmis, 2019-12-22:
I accidentally pressed the mid key on the developer side.  my car is a 2016 model 1.5 dc hatcback.the size of my screen is 8.7 inc. the screen turned 90 degrees.and also white spots appeared on the screen. thanks for your help

> shrlnm <notifications@github.com> şunları yazdı (22 Ara 2019 10:00):
> 
> Could you please explain your issue

shrlnm, 2019-12-22:
@engindalmis 
Ok
First try this
https://gitlab.com/py_ren/pyren/-/wikis/how%20to%20rlink2%20black%20screen

engindalmis, 2019-12-22:
the screen is straightened, but there are lines extending downwards on the left side. how can we fix this?

iPhone’umdan gönderildi

https://gitlab.com/py_ren/pyren/-/wikis/how%20to%20rlink2%20black%20screen

engindalmis, 2019-12-22:
> [cid:FBAAD7F8-1520-41AC-BD12-3D1A781A776D-L0-001]

iPhone’umdan gönderildi

Engin Dalmış <engindalmis26@hotmail.com> şunları yazdı (22 Ara 2019 12:52):

 the screen is straightened, but there are lines extending downwards on the left side. how can we fix this?

iPhone’umdan gönderildi

https://gitlab.com/py_ren/pyren/-/wikis/how%20to%20rlink2%20black%20screen

ouguena, 2020-03-20:
[image]

tyrogb, 2022-07-26:
Hello guys, can someone help me? I dont know exactly what I pressed maybe virgin mode, maybe the high mode. I tried with ddt4all that was slow for me for some reason cannot do anything with it, and then a tried pyron that didnt work also. 
[image]

AlMeu, 2022-07-26:
As it see you can not connect at all to your unit.
Do you have still sound in speakers?
The USB ports are still working?

tyrogb, 2022-07-27:
Yes everithing is working except the display.

AlMeu, 2022-07-27:
> First try this
> https://gitlab.com/py_ren/pyren/-/wikis/how%20to%20rlink2%20black%20screen
Try this

tyrogb, 2022-07-27:
I tried these none of the 4 works for me

AlMeu, 2022-07-27:
First be sure that you can connect to your unit and read it.
Without this you can't start do anything.

kuduro88, 2023-01-06:
> > Please help to exit this virgin mode.
> 
> e-mail?

Hello, could you help me too? I'm in the same situation :(

my email is: heldermartins3@gmail.com

misiek330, 2023-02-05:
Hey, you can help me because I tried to download the version from V2 to V3, unfortunately there is a black screen, I can't do anything email: misiek330@wp.pl

AlMeu, 2023-02-05:
@misiek330 I answerd you by email.

xalio1, 2023-05-10:
Hi all
I entered virgin mode and I have a black screen help me Renault ESPACE V 2015
I tried to do:
https://gitlab.com/py_ren/pyren/-/wikis/how%20to%20rlink2%20black%20screen
but it doesn't help
piotrekzajac1@wp.pl

shrlnm, 2023-05-11:
@xalio1 
https://www.drive2.ru/b/643019689764615344/

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 and one other....

closed 2023-06-02 by Furtif

octavius3000, 2025-02-06:
Hi, following the Russian tutorial I managed to get an R-Link 2 working again that was blocked by pressing the Virgin Mode button, however there are functions that have not been restored.
The radio is connected to R-Link 2 of 8 slots, installed in a Koleos 2018. The software version is 3.3.16.986.
The existing problems are the following:
1. Rear view camera: The rear view camera is not active until the vehicle is in motion. However, if I go to radio settings in the Parking Assistance, the rear view camera transmits images. With ddt4all I managed to return the image to its normal orientation because the le appeared inverted.
2. Air conditioning: The air conditioning buttons are not visible, so the air conditioning flow cannot be adjusted. I tried to activate it from ddt4all, in the ECU CONFIGURATION / ADAS PRESENCE / ACC screen. I was in the present. I changed it to Not Present and then back to Present but with no positive results.
3. Station capture devices: Do not function, do not detect obstacles.
4. Navigation: It does not have the map I updated. I am not too worried about this as it will probably be corrected with the installation of the patch and the new map.
Thank you in advance for your valuable help.
Sorry for my english.
Thank you.

Ocsi1980, 2025-03-28:
Hello! I reset the system from Virgin mode as described (megane 4 gt) but the message "emergency call to be checked" appears. There is no SOS button in the car. Could you help me how to turn it off?

alteadriver, 2025-04-04:
Could you help me, also pressed virgin mode, screen is black. Radio station and usb input shows on center display, but no sound. 
Tried pyren but it doesn't complete te macro.
Vin vf1rfa00158608662.