cedricp/ddt4all #305: error 7f2212

deadlyjigsaw, 2019-12-14:
[image]

i try to change to see the hour an temperature in the dashboard on my renault megane 3 2012
and i get this error every time it says there NODATA
can someone help me with this? thanks!

[image]

BarektE, 2019-12-14:
Hello, here, no one is helping. this will do the photo I send, not necessarily your screen.
[image]

shrlnm, 2019-12-14:
> can someone help me with this?

First you need to check that you are using an appropriate ECU file. 7F2212 means that your TDB not support the command you are using. It is usually because you selected ECU manually. 

Always use automatic ECU scanning/searching instead of manual selection. Otherwise you risk to damage your ECU by wrong commands.

deadlyjigsaw, 2019-12-14:
i have done also automatic ecu scanning and it is with green color that tdb ecu .... i believe is because of the elm interface that I have?

BarektE, 2019-12-14:
otomatik yapma benim işaret koyduğum tdb seç yaptıklarımı uygula derece ve saat gelecektir.

shrlnm, 2019-12-15:
> i have done also automatic ecu scanning and it is with green color that tdb ecu .... i believe is because of the elm interface that I have?

Not sure that it's ELMs guilty as so responses on other commands are there. I may check the scanning algorithm if you send me a log of scanning but I suppose that your TDB not support these parameters
 
Also not sure that the ECU suggested by @BarektE  will help as so it have it have the same LID numbers 0121 and 0122

BarektE, 2019-12-15:
> > Ben de otomatik ECU tarama yaptım ve yeşil renkli olan tdb ecu .... inanıyorum elm arayüzü nedeniyle olduğuna inanıyorum?
> 
> ELM'lerin suçlu olduğundan emin değilim, çünkü diğer komutlara verilen cevaplar orada. Bana bir tarama günlüğü gönderirseniz tarama algoritmasını kontrol edebilirim ancak TDB'nizin bu parametreleri desteklemediğini varsayalım.
> 
> Ayrıca, @BarektE tarafından önerilen ECU’nun 0121 ve 0122 LID numaralarına sahip olması için yardımcı olacağından emin değilsiniz.

Sayın @deadlyjigsaw shrlnm sizin bilginiz çok. Ancak ben uyguluyorum sorun olmuyor ve çalışıyor.

deadlyjigsaw, 2019-12-16:
Pe duminică,

 
 

  
ECU : CLUSTER_B95_L38_ph3_V1.92_20160427T140230.json
	@ 14:17:47.163
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Presence Airbag
		{"Presence Airbag": null}
	@ 14:17:47.241
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Présence MFD
		{"Pr\u00e9sence MFD": null}
	@ 14:17:47.303
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Siege chauffant
		{"Siege chauffant": "Pas de siege chauffant"}
	@ 14:17:47.350
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Type carrosserie
		{"Type carrosserie": "L"}
	@ 14:17:47.413
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Affichage freeshift
		{"Affichage freeshift": "Avec affichage Freeshift"}
	@ 14:17:47.460
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Presence Clock
		{"Presence Clock": null}
	@ 14:17:47.553
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Inhibition SBR
		{"Inhibition SBR": "Non inhibe"}
	@ 14:17:47.616
	Screen : Config Generale 2 & Odometrie	Request : DataRead.MPU_DRL
		{"MPU_DRL": "Avec Config MPU Day Running Light"}
	@ 14:17:47.663
	Screen : Config Generale 2 & Odometrie	Request : DataRead.AAP zone Display
		{"AAP zone Display": null}
	@ 14:17:47.725
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Presence LBCS
		{"Presence LBCS": "Sans LBCS"}
	@ 14:17:47.772
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Presence Module Conducteur
		{"Presence Module Conducteur": "Sans Module Conducteur"}
	@ 14:17:47.835
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Presence Temp_Ext
		{"Presence Temp_Ext": null}
	@ 14:17:47.882
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Gestion freeshift
		{"Gestion freeshift": "Avec gestion Freeshift"}
	@ 14:17:47.944
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Inhibition SBR arriere
		{"Inhibition SBR arriere": "Non inhibe"}
	@ 14:17:47.991
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Odometre general
		{"Odometre general": "204548"}
	@ 14:17:48.053
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Presence LDW
		{"Presence LDW": "Sans LDW (par defaut)"}
	@ 14:17:48.116
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Presence AHL
		{"Presence AHL": "Sans AHL (par defaut)"}
	@ 14:17:48.225
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Presence Pointer_Animation
		{"Presence Pointer_Animation": null}
	@ 14:17:48.272
	Screen : Config Generale 2 & Odometrie	Request : DataRead.MPU_SDO
		{"MPU_SDO": "Sans Config MPU Single Door Opening"}
	@ 14:18:33.743
	Screen : Config Service	Request : DataRead.Config Entretien_Service init
		{"Config Entretien_Service init.Revision Duration": null, "Config Entretien_Service init.Revision distance": null}
	@ 14:18:33.805
	Screen : Config Service	Request : DataRead.Config Entretien_ServiceDesappearenceDelay
		{"Config Entretien_ServiceDesappearenceDelay": null}
[... 962 lines cut ...]
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Présence MFD
		{"Pr\u00e9sence MFD": null}
	@ 14:28:45.541
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Siege chauffant
		{"Siege chauffant": null}
	@ 14:28:45.603
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Type carrosserie
		{"Type carrosserie": null}
	@ 14:28:45.666
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Affichage freeshift
		{"Affichage freeshift": null}
	@ 14:28:45.759
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Presence Clock
		{"Presence Clock": null}
	@ 14:28:45.853
	Screen : Config Generale 2 & Odometrie	Request : DataRead.Inhibition SBR
		{"Inhibition SBR": null}
	@ 14:28:45.947
	Screen : Config Generale 2 & Odometrie	Request : DataRead.MPU_DRL
		{"MPU_DRL": null}
	@ 14:28:45.994
	Screen : Config Generale 2 & Odometrie	Request : DataRead.AAP zone Display
		{"AAP zone Display": null}
	@ 14:28:46.056
	Screen : Config Gen

Valdu13, 2020-01-31:
bonjour j'ai un probleme urgent

Valdu13, 2020-01-31:
ddt4all sub function not supported (7F2212) clio 4 rs rlink 1

cedricp, 2020-01-31:
Il faudrait plus d'infos... Quelle fonction as tu essayé de paramétrer ? As tu fais une détection automatique ?
les fichiers logs pourraient servir aussi

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif