cedricp/ddt4all #307: > Maybe I was not perfectly clear. I have a car with ACC and I'm looking into reconfiguring the speedlimits.

bostu, 2019-12-21:
> Maybe I was not perfectly clear. I have a car with ACC and I'm looking into reconfiguring the speedlimits.
> Finally I managed to change the ACC speedlimits, however it only works temporarily. It seems the above referenced values are overwritten by an other ECU(?).
> After an engine stop and later restart the values are restored to the default ones.
> Originally ACC low limit is 50, below it is overridden to 30:
> [image]
> I'm still looking for some hints if any of you managed to permanently change the values.

hi gabr777,
did you have any luck with speed limits in ACC... I have espace 2016 too and also would like to set lower speed limits, like new cars have
and can you tell me where in ddt4all are these settings?
bx

_Originally posted by @bostu in https://github.com/cedricp/ddt4all/issues/126#issuecomment-568157159_

bostu, 2019-12-21:
I have espace 2016 and I would like to set lower speed limits, like new cars have
can you tell me where in ddt4all are these settings? or if this is even possible?
thanks
bx

tanerdogan, 2019-12-21:
I was looking same thing for normal cruise control speed limit but didnt found anything. I have Clio 4 EDC and want to use CC from start but there is also speed limit 30km/h for that too...

bostu, 2019-12-21:
I have ACC and automatic transmision, but I would like to have it from 0 km/h to drive in trafic jams

ouguena, 2020-03-20:
ddt4all
ACC > CMF1   Frontal_rader_v4b_0_light > Configuration_state_management_kmh 

ACCToACCOFFSpeedThresholdLow  40    =>>   20    edit
ACCOFFToACCSpeedThresholdLow  47    =>>   27    edit
MinimalSpeedRequestLow  50    =>>  30    edit

(***If it is lower than this value, bsw interference occurs.**)

and, 

ddt4all
ACC > CMF1   Frontal_rader_v4b_0_light > soft_Reset

ecu reset menu > softreset.

acc The low speed setting value does not change.

ouguena, 2020-03-20:
ddt4all
ACC > CMF1 Frontal_rader_v4b_0_light > Configuration_state_management_kmh

ACCToACCOFFSpeedThresholdLow 40 =>> 20 edit
ACCOFFToACCSpeedThresholdLow 47 =>> 27 edit
MinimalSpeedRequestLow 50 =>> 30 edit

(*If it is lower than this value, bsw interference occurs.)

and,

ddt4all
ACC > CMF1 Frontal_rader_v4b_0_light > soft_Reset

ecu reset menu > softreset.

acc The low speed setting value does not change.
[image]
[image]

bostu, 2020-03-20:
Thank you, I will try it

bostu, 2020-08-21:
Sorry for late answer, but I do this change yesterday. Everything works fine. thank you @ouguena. 
So if I understand right, it can go lower with speed limits?
 I could try but I am scared to do something stupid and then nothing works 😁. Can I destroy something if I try to go lower with speed limits or just it wont work and I will have to go up again with speed limits?
Thank you for answers

kasperronberg, 2021-03-15:
> Sorry for late answer, but I do this change yesterday. Everything works fine. thank you @ouguena.
> So if I understand right, it can go lower with speed limits?
> I could try but I am scared to do [...]

I have tried to change the values, and it works, but after you shut off, og turn on again, it changes back to old values.

Also it seams like the auto brakes don't work with low speed, only deaccelerate, so keep a foot on the brakes when you try 😉

Zaloznx, 2023-01-21:
Всем привет подскажите пожалуйста мужики можно ли активировать активный круиз контроль на рено сценик 4 2017года Доустановил кнопку на руль  а что дальше? кто поможет буду благодарен

Zaloznx, 2023-01-21:
нужны параметры активащии

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif