cedricp/ddt4all #308: Ecu tweaker is constantly stopped on Android system...

ssadicc, 2019-12-25:
java.lang.RuntimeException: An error occurred while executing doInBackground()
	at android.os.AsyncTask$3.done(AsyncTask.java:354)
	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
	at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
	at java.util.concurrent.FutureTask.run(FutureTask.java:271)
	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:764)
Caused by: java.lang.NullPointerException: Attempt to read from field 'long b.a.a.a.g$a.b' on a null object reference
	at b.a.a.a.g.c(Unknown Source:9)
	at b.a.a.a.g.b(Unknown Source:0)
	at b.a.a.a.b.a(Unknown Source:191)
	at org.quark.dr.canapp.MainActivity$a.a(Unknown Source:27)
	at org.quark.dr.canapp.MainActivity$a.doInBackground(Unknown Source:2)
	at android.os.AsyncTask$2.call(AsyncTask.java:333)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	... 4 more

ssadicc, 2019-12-25:
What's the wrong?

AndyIreland, 2019-12-28:
I just come across this myself today . I am pretty sure ecu tweaker worked fine on my xiaomi redmi note 5 pro .. but then that phone broke and my mobile phone provider then supplied me with a redmi note 6 pro and i tried loading ECU tweaker today and then it said it keeps stopping and came up with this:

java.lang.RuntimeException: java.lang.NoSuchMethodException: getSupportedDevices []
	at b.a.a.b.a.e.a(Unknown Source:91)
	at b.a.a.b.a.i.a(Unknown Source:7)
	at b.a.a.b.a.i.b(Unknown Source:2)
	at org.quark.dr.canapp.h.a(Unknown Source:27)
	at org.quark.dr.canapp.MainActivity.z(Unknown Source:175)
	at org.quark.dr.canapp.MainActivity.b(Unknown Source:0)
	at org.quark.dr.canapp.k.run(Unknown Source:18)
	at java.util.TimerThread.mainLoop(Timer.java:562)
	at java.util.TimerThread.run(Timer.java:512)
Caused by: java.lang.NoSuchMethodException: getSupportedDevices []
	at java.lang.Class.getMethod(Class.java:2068)
	at java.lang.Class.getMethod(Class.java:1690)
	at b.a.a.b.a.e.a(Unknown Source:5)
	... 8 more

[image]

ssadicc, 2019-12-28:
Firstly,
Thanks for answering
Yes I have this problem with my note 6 pro

java.lang.RuntimeException: An error occurred while executing
doInBackground()
at android.os.AsyncTask$3.done(AsyncTask.java:354)
at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
at java.util.concurrent.FutureTask.run(FutureTask.java:271)
at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
at
java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
at
java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
at java.lang.Thread.run(Thread.java:764)
Caused by: java.lang.NullPointerException: Attempt to read from field 'long
b.a.a.a.g$a.b' on a null object reference
at b.a.a.a.g.c(Unknown Source:9)
at b.a.a.a.g.b(Unknown Source:0)
at b.a.a.a.b.a(Unknown Source:191)
at org.quark.dr.canapp.MainActivity$a.a(Unknown Source:27)
at org.quark.dr.canapp.MainActivity$a.doInBackground(Unknown Source:2)
at android.os.AsyncTask$2.call(AsyncTask.java:333)
at java.util.concurrent.FutureTask.run(FutureTask.java:266)
... 4 more

What should we do about it?
Can you direct me?

28 Ara 2019 Cmt 23:51 tarihinde AndyIreland <notifications@github.com> şunu
yazdı:

smail2133, 2021-02-10:
No way ?

cedricp, 2021-02-11:
Can you give me the stacktrace with this version with debug symbols ?
https://drive.google.com/file/d/1mN_0ftiYFwC4l7XTWhFOJK2WdKgxz6BP/view?usp=sharing

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif