cedricp/ddt4all #309: R-Link dead after reset

Jeffreyvanesch, 2019-12-25:
Hi All,

I've tried to enable Android Auto for my Twingo by first updating the R-Link software through the SD card to the newest version (from the Renault website) When connection trough DDT4all with the OBDLink SX and a modified extension cable I'm able to access the can bus but unfortunately I pressed Hardware reset (reset hard) instead of soft.
After pressing the rlink rebooted and did not return to the radio screen it just stays all black and won't turn back on, this results in the R-link button lights not lighting up also the SD reader and USB are down at this point.

Refreshed the data in DDT and now recognizes only one ECU, (Identified ECU [MIU]@59 : BIC_A2_A3_v24_20180620T133153.json DIAGVERSION [20]SUPPLIER [DX2] SOFT [0620] VERSION [0000] {8898}) Manually selecting the Navigation ECU according to the guide on gps-rlink.com and from what I remember results in empty data fields @ the ecu side altough I'm able to fetch the old values when using read all.

Before it would show up as MFD_XXX_2012 and also had a (correct me if I'm wrong) Manually writing the values also doesn't work when expert mode is enabled as DDT doesn't seem to process them. 
Is there a way to recover the values in the ECU by unplugging the car battery or manual input since this is my fault or could this only be done by the dealer?

[image]
[image]
[image]
[image]

mnigten, 2019-12-25:
You have to find the ecu manually.  I never find the mfd automatic. Just scroll to navigation and select I think it was 4.1

Jeffreyvanesch, 2019-12-25:
Only have MFD 1.8, 4.3 or 5.4 under Navigation or under Audio: 2012 DAB 4.0, Radio 2012 v2.4, R0-13 v0.7 ,v2.0, v0.3, v0.4 OR Radio 2012 No DAB 2.0.
So with that I'd guess I'll try yo continue with the MFD 4.3 for now to see if I can fix it.

mnigten, 2019-12-25:
I think the version differences are minor changes. Mostly layout changes. 
There you should be able to set all changes except some of them because they are hard coded. On this side you will find some screenshots with some settings . They are mostly the same.

Jeffreyvanesch, 2019-12-26:
Tried it yesterday, Unfortunately no luck with this as the set values return empty after filling in and applying this.

Using [https://gitlab.com/py_ren/pyren]( Pyren) I'm able to fetch data from the Multimedia ECU again, The problem is that the 'Configuration' of the ecu or radio is empty / partly filled. Now the only way to reprogram the configuration is by using Can Clip OBD adapter and CLIP from Renault.

`ECU : 11173  BIC_X98_X87_0000_14_B
Screen : DE (STD_B) → DTC934155 COMP CONFIGURATION                                 1.DEF  ALIVE    

        1.DEF:Configuration new or incomplete.
`

I'd guess I'll take the Twingo to the garage and explain to them that the R-Link has crashed, Hoping that it will be fixed quickly.

shrlnm, 2019-12-26:
@Jeffreyvanesch 
What is in dump of bic? 

Check these values

22AC1C:62 AC 1C 16
22AC15:62 AC 15 0E
22AC1A:62 AC 1A FF FF FF FF

Jeffreyvanesch, 2019-12-27:
Looking at the latest dump

22AC1C:62 AC 1C 00 
22AC15:62 AC 15 0A
22AC1A:62 AC 1A FF FF FF FF

shrlnm, 2019-12-28:
On the configured device with VIN VF1AHB22551219604  they should be
22AC1C:62 AC 1C 16
22AC15:62 AC 15 0E

Jeffreyvanesch, 2019-12-28:
Cool, How can I write this to the ECU using PyREN or ?

KarelSvo, 2019-12-28:
> Twingo

Twingo 3 X07 Script 14

[image]

Jeffreyvanesch, 2019-12-28:
Great!, After programming this value the radio powers up again, @KarelSvo I'm wondering where you've got that from? 

Now the only issue left is to resolve the error message "Radio is not configured" with the yellow warning sign.

KarelSvo, 2019-12-28:
With ddt4all scan the ecus without an adapter cable.
And complete the rlink configuration.
Restore acoustic parameters with [pyren](https://gitlab.com/py_ren/pyren/tree/master/pyren/macro/rlink2).

Jeffreyvanesch, 2019-12-28:
Well I'm able to scan the ECU's but don't seem to be able to configure the parameters with DDT under, In expert mode (Einstein logo) I've set the configuration for Audio - Radio 2012 DAB v4.0 + Navigation MFD v4.6 but these values won't be saved to the ECU for some reason when pressing read all afterwards. (Returns empty values)
Using a soft reboot it directly returns to the not configured screen.

[image]

Looking at the log when into nav. MFD 5.4
ECU uses SDS 10C0
Sending ELM request :3E01
ELM response :  

It only sends requests but does noet get a response back in einstein mode.

As a precaution I did not try to restore the acoustic parameters with pyren since the scripts seem to be made for a rlink 2 and I've got a rlink 1.

KarelSvo, 2019-12-29:
> Well I'm able to scan the ECU's
I can't see your scan result.

shrlnm, 2019-12-29:
@Jeffreyvanesch give me your e-mail.

Jeffreyvanesch, 2019-12-29:
@shrlnm see profile

Jeffreyvanesch, 2019-12-29:
@KarelSvo 

> I can't see your scan result.

The following ECU's are beeing detected using the OBD adapter directly:
- Identified ECU [Power steering]@04 : X07_EPS_V3.0_20170811T134219.json DIAGVERSION [4]SUPPLIER [373] SOFT [D201] VERSION [100A] {2891}
- Identified ECU [BCM/UCH]@26 : T4_BCM_BIS_DDT2000_SW7_6_20131125T093324.json DIAGVERSION [9]SUPPLIER [021] SOFT [0007] VERSION [0300] {882}
- Identified ECU [UPC-EMM]@27 : EMM_EDISON_DDT2000_SW12_V1_6_20140407T162219.json DIAGVERSION [8]SUPPLIER [414] SOFT [C121] VERSION [3030] {8338}
- Identified ECU [ECM/Injection]@7A : ME17.920_1780_8E0_A50_B40_V1.0_20190125T113507.json DIAGVERSION [68]SUPPLIER [037] SOFT [1780] VERSION [8D00] {6731}
- Identified ECU [HVAC]@29 : ClimBoxGen2_v29_20190628T153255.json DIAGVERSION [4]SUPPLIER [740] SOFT [D406] VERSION [1408] {1361}
- Identified ECU [DCM Renault]@77 : TCU_GEN2_V2.0_20140423T130205.json DIAGVERSION [8]SUPPLIER [284] SOFT [0367] VERSION [0301] {8446}
- Identified ECU [Airbag]@2C : RSAT4_ACU_eng_v15_20150511T131328.json DIAGVERSION [4]SUPPLIER [AMR] SOFT [1630] VERSION [1630] {6127}
- Identified ECU [Cluster/TDB]@51 : Cluster_X07_ICE_V4.1_20181212T124942.json DIAGVERSION [4]SUPPLIER [096] SOFT [0007] VERSION [0700] {227}
- Identified ECU [FCAM]@62 : LDW_ECU_X07_X62_v2.7_20170908T155816.json DIAGVERSION [8]SUPPLIER [AFK] SOFT [0201] VERSION [0000] {5554}

With the modified cable:
- Found ECU [Navigation] (not perfect match) :MFD v5.4 DIAGVERSION [20] SUPPLIER [329] SOFT [2D3C] VERSION [412D instead 452F]
- Identified ECU [Audio]@13 : Radio_2012_DAB_v4.0_20151012T163337.json DIAGVERSION [116]SUPPLIER [BT2] SOFT [0400] VERSION [0350] {2063}
- Identified ECU [MIU]@59 : BIC_A2_A3_v24_20180620T133153.json DIAGVERSION [20]SUPPLIER [DX2] SOFT [0620] VERSION [0000] {8898}

With [audio] Radio_2012_DAB_v4.0 and [MIU] listed in green, MFD 5.4 is listed in red.

shrlnm, 2019-12-29:
@Jeffreyvanesch try this macro with pyren
https://cloud.mail.ru/public/4EEo/2qdcHjXSd

Jeffreyvanesch, 2019-12-29:
Well it seems that with the last programming session the values were accepted just by 'exiting the car and locking the door' method the 'Not Configured' message disappeared once re-entering the car after about 30 minutes.
Now I've got to enter the radio security code after a few seconds (which is stated somewhere on the paperwork of the car and need to resolve the series of speaker beeps which I read elsewhere in the issues section.

closed 2019-12-31 by Jeffreyvanesch

kzt1979, 2020-04-10:
Dear all
I have the same problem , 
Please can you help me

shrlnm, 2020-04-11:
@kzt1979  
What type of reset did you do?  
What LGU version? 
What MICOM (boot) version?
e-mail address?

kzt1979, 2020-04-11:
@shrlnm 
thank you for your reply
I Did software reset.
I lost all configuration, but I restored most of them. At the end the I  had a message saying "unconfigured device please add eol settings"
my email is  kzt1979@yahoo.com
[image]

mnigten, 2020-04-12:
Please read this subject. I managed to get it working but i had still problems with radio not configured. The eol message was resolved. 
https://github.com/cedricp/ddt4all/issues/92#issuecomment-457928308

nottiw, 2020-04-17:
Dear All,
I have the same problem with yellow triangle. I've bought all R-Link navigation components from accident cars (different, not one car) and installed them on my Laguna 3 car, where originally was old Carminat DVD navi. And after first switching on, I saw this screen. I tried to check through DDT4all what happened and saw that the Radio has most of the empty data, but I can change and save it,. However, MFD also has no data, but I can't save it again.

[image]

seb0031, 2020-07-05:
> > Twingo
> 
> Twingo 3 X07 Script 14
> 
> [image]

Bonjour, je suis avec le message d'erreur "eol settings" pouvez vous m'aider j'ai une zoe de 2013. j'ai voulu mettre a jour android auto avec ddt4all. depuis je peu plus rien faire. help me help me

cordialement

seb0031, 2020-07-05:
Bonjour, 

je suis dans la même galère, avec le message d'erreur "eol settings". pouvez-vous m'aider ? j'ai une zoe de 2013. j'ai voulu mettre a jour android auto avec ddt4all. depuis je peu plus rien faire. help me help me

cordialement

seb0031, 2020-07-05:
> @shrlnm
> merci pour votre réponse
> J'ai réinitialisé le logiciel.
> J'ai perdu toute configuration, mais j'en ai restauré la plupart. À la fin, j'ai reçu un message disant "appareil non configuré, veuillez ajouter les paramètres eol"
> mon e-mail est [kzt1979@yahoo.com](mailto:kzt1979@yahoo.com)
> [image]

bonjour, as tu une solution ?

seb0031, 2020-07-05:
> Salut à tous,
> j'ai le même problème avec le triangle jaune. J'ai acheté tous les composants de navigation R-Link de voitures accidentées (différentes, pas une seule voiture) et les ai installés sur ma voiture Laguna 3, où se trouvait à l'origine le vieux Carminat DVD navi. Et après la première mise sous tension, j'ai vu cet écran. J'ai essayé de vérifier via DDT4all ce qui s'est passé et j'ai vu que la radio contient la plupart des données vides, mais je peux les modifier et les enregistrer. Cependant, MFD n'a pas non plus de données, mais je ne peux malheureusement pas les réécrire.
> 
> [image]

Bonjour, as tu trouvé la solution ?

AndyIreland, 2021-02-27:
> Well it seems that with the last programming session (With DDT4all and without the scripts) the values were accepted just by 'exiting the car and locking the door' method. Also, the 'Not Configured' message disappeared once re-entering the car after about 30 minutes.
> After this I needed to enter the radio's security key to have full access to the radio again and stop those annoying beep tones trough the speakers.
> 
> Any parameters left that I need to check with DDT or Pyren?

@Jeffreyvanesch  - hello . i done same thing today , reset my rlink - bad move got the yellow triangle EOL. Just went out to car inputted the parameters in DDT4all and then pressed 'hard reset' then all of a sudden beeps out of the speakers in rotation. Its 11pm so late and didnt want to wake up neighbours so I have disconnected the 12v and will try tomorrow - do you reckon when I re-connect the triangle might be gone then is that what happened with yours ? - and the secrity code?, you got that from renault did you or in book or in my renault app? 

Thanks.

Jeffreyvanesch, 2021-02-28:
Hey @AndyIreland,
Think that's a bad move, disconnecting the 12v results in the need to reenter the Security Code, and setting the internal clock because the unit has no backup battery like a PC, wasn't it possible to simply switch off the radio?
The triangle won't be gone after reconnection since the unit is in locked state if I remember correctly. The only way to get rid of it is to reprogram the radio ECU with Pyren.

Since you mention you're getting the rotating beeps it only needs the security code which usually can be found on the sales contract of the car or on the radio unit itself.
In case you don't have the sales contract, or it's not mentioned then the only way to get it fixed is via your garage as they will need to program the ECU with their CLIP software (which only works with the special dongle).

AndyIreland, 2021-03-04:
> Hey @AndyIreland,
> Think that's a bad move, disconnecting the 12v results in the need to reenter the Security Code, and setting the internal clock because the unit has no backup battery like a PC, [...]

Thanks for your reply, i managed to fix it myself over the weekend with a TomTom.000 file on the SD card and filled in all the parameters manually and the vin code of the car including CRC  then hard reset of radio, then security code popped up and I got the security code from the my.renault android app 

Thanks.

luissilva1971, 2022-04-02:
Hello. I am trying to replace original non bluetooth radio on my megane 3 2010 with GPS (carminat TOMTOM non live) with another original radio but with bluetooth function i bought on a scrapyard. The car the radio came from did not have GPS and it is from same year. Through a friend of mine i could change the conf of the radio to GPS (CLIP), but he did not get how to change the GPS config for the radio (it simply do not appear on CLIP screen!!!). Well 
I have been studing the process (DDT4all, PyREN ..) i from what i been reading i need to change parameter R1_08 on NFA v0.6 ECU to R2_08 (is it just that?, or it there anithing else?). 
I scanned the ECUs with ELM327 wifi and a modified cable for scanning CAN2 with PyREN and i have made a backup / dump of NFA throught pyREN. With DDt4all i have also achived to change VIN/CRC of the radio to the one it is on my car. Now, i would like to know if you can explain step by step would can i change the parameter on DDT4all. I have scanned the car and it shows 3 ECU, 2 on green (radio and another BIC ..) and a ITT_navigation on red. I know i can manually select NFA V0.6, but i have doubt how i can change paremeter correctly and i would like that you can clarify me. First go on conf W/R, go to radio type and change R1_08 to R2_08; 2nd, set "expert mode" and it "Write conf" to write to ECU, right? After this does the bar radio should appear (in order to see on the GPS the radio and write the pin code to unlock it)? or after the write conf i need to click on "hard reset" on reset menu? I would be forever thankful to all that can help me. Kind regards.

vraug666, 2022-06-21:
Hello, I have a r link after the reset does not want to run, black screen. Via the modified cable both DDT4all and Reno link cannot find computers from navigation or radio. Can somebody help me?

AndyIreland, 2022-06-21:
> Hello, I have a r link after the reset does not want to run, black screen. Via the modified cable both DDT4all and Reno link cannot find computers from navigation or radio. Can somebody help me?

does screen light up even if its black or blank, can you see the backlight of the screen. and try pressing home button 5 times . and try disconnecting 12v battery under bonnet for 10 mins then connect back up (if it does start up you may need to enter the r-link security code)
and you are sure the rewired lead or adaptor is wired properly yep?

criss717, 2024-02-19:
> Ekzt1979 What kind of reset did you do? Email address?

Hello, this same thing happened to me on my Renault Megane 3, please help, the screen remains black and it only scans the BIC with the modified cable, how do I see that from the BIC dump? my email is mono-717@hotmail.com

johnkar87, 2025-02-13:
> > Twingo
> 
> Twingo 3 X07 Script 14
> 
> [image]

can I do it for rlink 2 because is total black and resetting every 15sec . I know why is reseting because when is reseting blink green color (like eco mode of engine) and then I have blue color

AlMeu, 2025-02-13:
@johnkar87 here is speaking about R-Link 1.
If your R-Link 2 behave like you say, need change it only.

bcd94, 2026-08-18:
Please can anyone help me! Mine rlink is on messagem radio not configured...
I connected pyren ddt4all renolink and it seems all configured..
Piren bot not working... 
I have 
Data index="28" name="Radio_reception" ReadRequest="DataRead.Radio_reception" WriteRequest="DataWrite.Radio_reception" Value="00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000">

Please can anyone help me! Thank you

mishu28nmv, 2026-08-19:
If you provide your VIN, maybe it will work with pyren dumps/scripts. I can download that for your VIN number.

bcd94, 2026-08-19:
Good morning sir, 
Yes mine vin is VF12RFL1H49681807

[VF12RFL1H49681807.zip](https://github.com/user-attachments/files/31210133/VF12RFL1H49681807.zip)

mishu28nmv, 2026-08-19:
Here are your files for your VIN:

[VF12RFL1H49681807_Repair.zip](https://github.com/user-attachments/files/31217369/VF12RFL1H49681807_Repair.zip)

bcd94, 2026-08-20:
Hi misbu28nmv, it worked! Thank you, how can i pay you! Thank you

mishu28nmv, 2026-08-21:
No need to pay! I'm glad you did it!