cedricp/ddt4all #310: AttributeError: 'module' object has no attribute 'QWebEngineView'

coolkom, 2019-12-28:
Hello,
I have installed the program on a mac (macosx 10.15.2). I have purchased a USB module EML 327 and everything seems fine except anytime I click on the Edition Mode or Connected Mode I get this error:

Using custom DDT database
Initilizing ELM with speed 38400...
Trying to open port/dev/cu.usbserial-AQ00GI1Y @ 38400
1514 loaded ECUs in database.
Traceback (most recent call last):
  File "ddt4all.py", line 1434, in <module>
    w = Main_widget()
  File "ddt4all.py", line 294, in __init__
    self.docview = webkitwidgets.QWebEngineView()
AttributeError: 'module' object has no attribute 'QWebEngineView'

Please could you help me with this? 
Thanks,
Coolkom.

KarelSvo, 2019-12-29:
https://brew.sh/
brew install pyqt5

coolkom, 2019-12-29:
Thanks KarelSvo.
The issue is that, even with qt5, I get an error:

Traceback (most recent call last):
  File "ddt4all.py", line 24, in <module>
    import PyQt4.QtGui as gui
ImportError: No module named PyQt4.QtGui

So, am I the only one getting this with a mac?

KarelSvo, 2019-12-29:
brew install pyqt4

coolkom, 2019-12-29:
"brew install pyqt4" does not work (anymore). I install pyqt4 with "brew install cartr/qt4/pyqt@4" but this does not solve the problem.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836