cedricp/ddt4all #311: Manual requests through CMD

sirbig, 2020-01-02:
Hello everyone,

My question is quite simple: Is it possible to make manual request through CMD instead of using the GUI?

Thanks.

KarelSvo, 2020-01-03:
[img]https://github.com/cedricp/ddt4all/blob/master/icons/hex.png[/img]

sirbig, 2020-01-03:
> https://github.com/cedricp/ddt4all/blob/master/icons/hex.png
> this option

But that's through the GUI!
I want to do it by CMD Prompt.

KarelSvo, 2020-01-03:
Why do you take ddt4all when you want to work with a terminal program?

sirbig, 2020-01-03:
I want to send manual requests and receive the data to process on my own application.
PyRen should have this option since its a more command prompt driven app but it has no manual requests option.

shrlnm, 2020-01-03:
> PyRen should have this option since its a more command prompt driven app but it has no manual requests option.

mod_term.py from pyren and macro may helps.

sirbig, 2020-01-03:
> > PyRen should have this option since its a more command prompt driven app but it has no manual requests option.
> 
> mod_term.py from pyren and macro may help.

@shrlnm Once again, you are a life saver!
That's exactly what I was looking for, thank you so much!

Also, if anyone is interested in a .NET solution for OBD communications, here is a github of a really interesting project: https://github.com/DarthAffe/OBD.NET

sirbig, 2020-01-03:
@shrlnm I'm trying to execute a commands using "mod_term.py" but it always says "WRONG RESPONSE"...

I've set the target, rx and tx. I try to use a HEX command but it doesn't work.
Example (\macro\init.txt): **$addr = 51**
                                          **$txa  = 743**
                                          **$rxa  = 763**

I initialize it with: **%comspec% "/K .\..\Python27\python.exe .\mod_term.py -p COM5 --demo"**
Screenshot of the terminal attached.
[image]

shrlnm, 2020-01-03:
> but it always says "WRONG RESPONSE"...

Look at the end of prompt. It ends with ELM and it mens that you did not init the protocol. You may do it by "can500" for example.

i.e in your case your macro should start int this way

$addr = 51
can500
10C0
222201

Most advanced example of macro is there

https://gitlab.com/py_ren/pyren/blob/master/pyren/macro/qr25/idle_speed.cmd

sirbig, 2020-01-03:
Still it gives the same error and when I set the session it gives **"ODD ERROR"**
Maybe it's because I'm in demo mode instead of retrieving live data from ECU?

shrlnm, 2020-01-04:
> Maybe it's because I'm in demo mode instead of retrieving live data from ECU?

Yes it is.
For debugging purpose in demo you may use option --debug. In this case the terminal will use './dumps/term_test.txt' in format of usual dump for emulating ECU responses and file './macro/test/test.cmd' as a source of commands

sirbig, 2020-01-04:
I guess my PyREN version is too old and there is no --debug option to start mod_term.py, I will update with the one on gitlab that you sent me.
However, I tested mod_term.py with live data from my Clio IV and it worked fine.

Thanks again for all your help @shrlnm !

closed 2020-01-04 by sirbig