cedricp/ddt4all #312: Clio 4 EMM Replacement

ecutechnology, 2020-01-05:
Hi, could someone tell me how to change VIN in EMM using ddt4all please. I am replacing with an identical used module.

Diagnosed faulty by dealer, no main headlights when engine running.

Clio 4 2016

Thanks,

Luke

KarelSvo, 2020-01-06:
If not integrated in the screen, go to the list to Manual request.

[image]

Siqy, 2022-05-20:
Hi all,

Sorry for opening an old issue but I have a question that follows up this issue.

You only have to fill in the complete VIN number of your car in the "Value" table on the right side of the table "key" "V.I.N. in ASCII"? nothing more nothing less?

Thanks in advance!

Joël

Siqy, 2022-05-20:
Just installed the second hand EMM out a captur 2013 with the same codes as mine Clio 2013. Works without programming! Did not program the VIN of my car, read out the VIN or the module and it is different from my car.

on the other hand, DDT4all does not make a full match of the module for programming purposes (2019 database). Should I still program my cars VIN in the module or let it be?
The question above still remains.. answer would be nice!

Thank you in advance.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif