cedricp/ddt4all #313: Clio IV main buzzer

sirbig, 2020-01-08:
Hi everyone,

Does anyone know how to test/read the main buzzer?
So far I can only find it for the Instument Panel buzzer.

I searched in every ECU I have and no luck finding this fella...
[image]

premultiply, 2020-01-08:
Do you mean the main horn?

sirbig, 2020-01-08:
> Do you mean the main horn?

Yes.

AndyIreland, 2020-01-19:
do you mean the alarm 'blip blip' sound as you walk away and car locks?

sirbig, 2020-01-19:
> do you mean the alarm 'blip blip' sound as you walk away and car locks?

No, the main horn, the one that you use on your steering wheel when someone does something stupid on the road!
I think "main horn" is pretty clear, lol.

AndyIreland, 2020-01-20:
ah right I get you now. Thanks for clarifying it LOL ... but no, I havent seen a setting in PyREN for testing the horn or the steering wheel button soz.

Mikescotland, 2021-04-22:
Clio IV (same as Clio III) has horn connected directly to the wheel buttons. It doesn't go via any can line (simple electrics). Therefore you can't control / check horn via obd.

sirbig, 2021-04-22:
> Clio IV (same as Clio III) has horn connected directly to the wheel buttons. It doesn't go via any can line (simple electrics). Therefore you can't control / check horn via obd.

My solution was adding a RF Transmitor with a remote control to activate/deactivate the horn.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif