cedricp/ddt4all #316: Enabling ACC (adaptive cruise control)

robertalks, 2020-01-15:
Hello, I haven't tested this nor yet checked, but I have a Mega IV from 2016 which has the radar on the front, but I don't actually have adaptive cruise control (the button on the steering wheel). Is it possible somehow possible to enable ACC and add the button? I know is a strange question, but I'm guessing because I have the front radar it should be only a software thing and of course replacing the buttons on the steering wheel.

Any ideas?

Banana009, 2020-01-26:
Unfortunately, we don't have a procedure to activate ACC even if you have a radar. The problem is, you don't have 5 butons on the  steering wheel.

robertalks, 2020-01-27:
Can those be added? I mean, the steering wheel can be replaced or just the buttons.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif