cedricp/ddt4all #318: Change Time the Interior Light is on?

AndyIreland, 2020-01-20:
hi . in my Clio 4 there is a setting in my BCM/UCH ECU for  INTERIOR_LIGHTS_TM_TP and it is set for 20seconds in my car. I would like to increase to 30seconds or even 60seconds if I can but in the DDT4all window it loooks like it only reads the parameter and cannot right to the ECU - is this right? , is there no way I can change the on time of my interior lights through DDT4ALL? 

[image]

BarektE, 2020-01-20:
[image]

AndyIreland, 2020-01-20:
thanks - will give it a go cheers

AndyIreland, 2020-01-20:
Thanks for your suggestion 
all I got was this - but its not the TM and didnt have  'seconds' after it  - so I think it was just for enabling interior light (which it already is) 
[image]

I then got hopeful when i saw this 
[image]

because I could actually change the figure in the box from 20 to 60 ... but it was short lived because when I press 'execute' it goes back to 20 seconds. - mind you thats because its 'read only' I suppose .

BarektE, 2020-01-21:
I will try with Megane
I will inform if it happens

BarektE, 2020-01-21:
[image]

BarektE, 2020-01-21:
[image]

org07, 2020-03-19:
Hello Andy Ireland Clio 4 right headlight does not light on my car, not electrical work, how to fix it with software work ddt4all.

AndyIreland, 2020-03-19:
hello @org07 - i would say that the software wont fix it, i would say either loose wire, or blown fuse /relay or blown bulb

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif