cedricp/ddt4all #319: Renualt Talisman, How to activate the high beam function with Pure Vision LED headlamps?

nyaben, 2020-01-22:
Hello, everybody. the Pure Vision, LED head lamps is an optional thing in Korea. I bought and installed the Pure Vision secondhand recently, but the high beam function is not working properly. High beam can only be turned on when the low beam is turned on. 

can anyone suggest me a solution? thank you.

shrlnm, 2020-01-22:
@nyaben Hello, what is your vin and donors vin number

nyaben, 2020-01-23:
> @nyaben Hello, what is your vin and donors vin number

Geez, do I need both of information? ofcourse, it's super easy to check my VIN but the donor's one? that is not possible in my situation. I bought the Pure Visions from local auto repair shop and I'm pretty sure they are not tracking the number.

AndyIreland, 2020-01-23:
On my Clio 4 with DDT4all I changed this parameter :
either Dual_bulbs or  Bi_Function you have to adjust the option to '0' or '1' accordingly . then when you push your hi beam stalk  on the steering wheel the low beam light will extinguish and the high beam only will be lit. pull the hi beam stalk and the high beam lamp extinguishes and low beam lamp will come on . - change the parameter again and you can make it so hi beam lamp comes on along with the dipped beam lamps at the same time 

[image]

shrlnm, 2020-01-23:
> but the donor's one?

Ok. Your vin only maybe enough

shrlnm, 2020-01-23:
I've just checked. It's about 50 different parameters for random VIN number between LAHALO and LEDH2. Talisman also may have LEDH3. It is why I asked about donors vin.

nyaben, 2020-01-24:
> On my Clio 4 with DDT4all I changed this parameter :
> either Dual_bulbs or Bi_Function you have to adjust the option to '0' or '1' accordingly . then when you push your hi beam stalk on the steering [...]

You have my thanks. I could reach to the address according to the screenshot you attached and change several options from Halogen to LED. it was not the solution but I could solve a little problem that I didn't mention.
Anyway, there were some options about high beam function on the list but for some reason I couldn't make change the value for them. Weird.. They were commonly written as 'NO DATA' in red blanks. Any idea of them?

[image]

nyaben, 2020-01-24:
> I've just checked. It's about 50 different parameters for random VIN number between LAHALO and LEDH2. Talisman also may have LEDH3. It is why I asked about donors vin.

Whooah 50, that's huge number. That must mean I gotta try something in 50 different way, right? If you're going to tell me how, I will try surely. Thank you for giving me hints.

shrlnm, 2020-01-24:
@nyaben it means that you have to change about 50 parameters in the USM for tune it for working with LED head light. 

The parameters below are for tune  random Talisman, originally made with halogen lamps, to work with LED type 2. 

These parameters from USM_CMF1_SED_C1_2_170316_20170317T113830.xml

--< LEDH2
--> LAHALO

------------------------------
E_U_LowBeamRightOffset
--< 0x00
--> 0x28
------------------------------
E_B_Hbeam_OC_Inhibition
--< 0x01
--> 0x00
------------------------------
E_Pct_MAN_SWI_INDEX_TAB_3
--< 0x00
--> 0x6e
------------------------------
E_Pct_MAN_SWI_INDEX_TAB_2
--< 0x00
--> 0x43
------------------------------
E_Pct_MAN_SWI_INDEX_TAB_1
--< 0x00
--> 0x14
------------------------------
E_Pct_MAN_SWI_INDEX_TAB_5
--< 0x00
--> 0xca
------------------------------
E_Pct_MAN_SWI_INDEX_TAB_4
--< 0x00
--> 0xa0
------------------------------
E_B_Cornering_Inhibition
--< 0x00
--> 0x01
------------------------------
E_B_LowBeamWithoutPWM_sty_inv
--< 0x00
--> 0x01
------------------------------
dtc_INH_O_LIGHT_HBEAM_LH_SCS_OC
--< 0x01
--> 0x00
------------------------------
E_K_MAN_SWI_FIL_COEF
--< 0x00
--> 0xce
------------------------------
E_Pct_NOM_PWM_OFS
--< 0x1fd6
--> 0x0feb
------------------------------
E_B_LowBeamWithoutPWM
--< 0x01
--> 0x00
------------------------------
dtc_INH_O_LIGHT_LBEAM_RH_OC
--< 0x01
--> 0x00
------------------------------
E_B_HighBeamWithoutPWM
--< 0x01
--> 0x00
------------------------------
E_E_LVL_CFG
--< 0x01
--> 0x03
------------------------------
dtc_INH_LI_LED_1_Failure
--< 0x00
--> 0x01
------------------------------
E1d_HighBeam_Law_Y 1
--< 0x5a
--> 0x51
------------------------------
E1d_HighBeam_Law_Y 3
--< 0x4b
--> 0x44
------------------------------
E1d_HighBeam_Law_Y 2
--< 0x52
--> 0x44
------------------------------
E1d_LowBeam_Law_Y 2
--< 0x52
--> 0x44
------------------------------
E1d_LowBeam_Law_Y 3
--< 0x4b
--> 0x44
------------------------------
E1d_LowBeam_Law_Y 1
--< 0x5a
--> 0x51
------------------------------
dtc_INH_O_LIGHT_LBEAM_RH_SCS_OC
--< 0x01
--> 0x00
------------------------------
E_E_LowBeamType
--< 0x01
--> 0x00
------------------------------
E_U_MinimumVoltageByLight
--< 0x78
--> 0x88
------------------------------
E_B_PassingInhibition
--< 0x01
--> 0x00
------------------------------
E_Pct_MAN_SWI_INDEX_BAND
--< 0x00
--> 0x0f
------------------------------
E_U_HighBeamRightOffset
--< 0x00
--> 0x50
------------------------------
E_Pct_RR_SSR_INIT_RGE_L
--< 0x42
--> 0x26
------------------------------
dtc_INH_O_LIGHT_HBEAM_RH_SCS_OC
--< 0x01
--> 0x00
------------------------------
E_U_MAN_SWI_DIAG_ELEC_SCP
--< 0x0000
--> 0x782a
------------------------------
E_U_MAN_SWI_DIAG_ELEC_SCG
--< 0x0000
--> 0x0225
------------------------------
E_U_HighBeamLeftOffset
--< 0x00
--> 0x28
------------------------------
dtc_INH_O_LIGHT_LBEAM_LH_OC
--< 0x01
--> 0x00
------------------------------
dtc_INH_O_LIGHT_HBEAM_RH_OC
--< 0x01
--> 0x00
------------------------------
dtc_INH_O_LIGHT_LBEAM_LH_SCS_OC
--< 0x01
--> 0x00
------------------------------
E_U_LowBeamLeftOffset
--< 0x00
--> 0x14
------------------------------
E_FT_SSR_ATT_COEF
--< 0xdaef
--> 0x0000
------------------------------
dtc_INH_LI_LED_2_Failure
--< 0x00
--> 0x01
------------------------------
E_D_WHEEL_HDLMP_DIST
--< 0x59
--> 0x60
------------------------------
E_E_HighBeamType
--< 0x01
--> 0x00
------------------------------
dtc_INH_O_LIGHT_HBEAM_LH_OC
--< 0x01
--> 0x00
------------------------------
dtc_INH_Levellizer_Sensor_Initialisation
--< 0x00
--> 0x01
------------------------------
E_Ts_MAN_SWI_INDEX_TIM
--< 0x00
--> 0x08
------------------------------
E_K_Cor_Dimming_step_nb
--< 0x0f
--> 0x05
------------------------------
E_RR_SSR_ATT_COEF
--< 0xc290
--> 0x0000

nyaben, 2020-01-25:
> @nyaben it means that you have to change about 50 parameters in the USM for tune it for working with LED head light.
> The parameters below are for tune  random Talisman, originally made with halogen lamps, to work with LED type 2.
> These parameters from USM_CMF1_SED_C1_2_170316_20170317T113830.xml

[image]

Big thanks, shrlnm!I truly appreciate it. but maybe I'm too amateurish to understand your writing. I could find some items above from the DDT4all program like 'E_E_LowBeamType', but not for some like 'dtc_INH_O_LIGHT_LBEAM_RH_SCS_OC'. and all the values you told me starting with '0X..'.. getting confused.

shrlnm, 2020-01-25:
Yes, it may be a bit confusing. It is better to do such things automatically. I asked your vin not out of  curiosity. I may do a pyren macro for you witch may change all of these parameters. 
You may give me your email if you don't want to show your vin in public.

nyaben, 2020-02-01:
Guys, I finally found how to do ; UPC-EMM > USM_CMF1_... > LIGHTS > LIGHTS CONFIGURATION > E_B_PassingInhibition , change the value from 0 to 1. That's it. 
Actually only passing function (not the high beam function when you push the lever back) works but I satisfied.

Thank you all for helping me and especially shrlnm! You tried to help me up readily to the end!

closed 2020-02-01 by nyaben

Banana009, 2020-02-29:
I also plan to replace classic halogen lights with  Pure Vision LED headlamps on my Grand Scenic IV. 
I'm a little confused about this discussion, can you advise me what all the parameters need to be change?
Is it necessary to change cabling during installation?
Thanks for all the advice.