cedricp/ddt4all #320: How to run on MACOSX ?

Virallata, 2020-01-23:
I Installed python, Pytq4 and 5, and i stil got this error

Kaues-MacBook-Air:ddt4all-master kaue$ /usr/bin/env python ddt4all.py
Traceback (most recent call last):
  File "ddt4all.py", line 24, in <module>
    import PyQt4.QtGui as gui
ImportError: No module named PyQt4.QtGui

thx

goldenlindt, 2020-08-14:
Hey,
did you find a solution for your problem? I'm currently also struggeling to get ddt4all running on my mac

Anxiro, 2020-08-25:
I got that problem finally fixed by running, and please continue if you see errors:

```
pip3 install PyQt5
pip3 install PyQtWebEngine
pip3  uninstall PyQt5
pip3  uninstall PyQt5-sip
pip3  uninstall PyQtWebEngine

pip3 install PyQt5
pip3 install PyQt5-sip
pip3 install PyQtWebEngine
````

Yes, I removed them and re-added them. Then I did, inside the ddt4all-folder:

```
mv importlib importlib_bak
python3 ddt4all.py
```

Which actually made the app run. I do have issues on my Mac _(MacBook Pro (16-inch, 2019), 2,4 GHz 8-Core Intel Core i9)_ with the WiFi connection. I do see the OBD blink as it does fetches a connection:

```
Initilizing ELM with speed 38400...
Trying to open port192.168.0.10:35000 @ 38400
Port closed
Trying to open port192.168.0.10:35000 @ 38400
Port closed
Trying to open port192.168.0.10:35000 @ 115200
Port closed
Trying to open port192.168.0.10:35000 @ 230400
Port closed
Trying to open port192.168.0.10:35000 @ 57600
Port closed
Trying to open port192.168.0.10:35000 @ 9600
Port closed
Trying to open port192.168.0.10:35000 @ 500000
Port closed
****************************************
*       Connection to ELM was lost
Port closed
```

But that's another story.

Timmytubby, 2021-04-19:
Did mine with python launcher. Works like a charm, only reading live in the vehicle shows up as bright yellow. Change screen settings to black and gray, work around 😂

makayk, 2021-04-22:
Thanks Anxiro, it is working, have fun! :)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif