cedricp/ddt4all #322: Is DDT4All usable on a brand new ECU? (Easy Link / A-IVI)

michaelmcneela, 2020-01-26:
Howdy,

I intend to purchase a brand new A-IVI ECU ("radio") from Renault– will it be configurable via DDT4All immediately (once I connect power, and CAN), or will it first require firmware flashing etc? (using what other tool?)

Thanks

KarelSvo, 2020-03-14:
I have two A-IVI dump, 7 "and 9.3".
Are you doing a virgin dump for comparison?

michaelmcneela, 2020-03-14:
@KarelSvo Ooh, I'd _love_ to see the dumps, if you're willing to share them.

I got a used A-IVI unit in the end (from a Clio with 9.3"), because I didn't think I'd be able to configure a virgin unit, either because;
- the unit would have no firmware (and so obviously _wouldn't_ respond to ddt4all), 
- or it _would_ have firmware (and respond in ddt4all), but I wouldn't have usable acoustic parameters to set.

In theory, the A-IVI unit I have is ready to use, but– **my current problem, stopping me from continuing**: the unit, as soon as it is given power, seems to go into a low-power sleep mode– waiting for a wake-up signal via CAN, and doesn't respond to ddt4all at all.

[image]

From the A-IVI ECU XML file; the **Vehicle_Information.PowerPlatform** setting appears to configure the wake-up behaviour, and I believe it's likely currently set to **V-CAN wake-up C1A (Renault C1A)**.

If I was able to wake my unit _just once_, for long enough for ddt4all to communicate with it, I should be able to set the **Vehicle_Information.PowerPlatform** value to **ACC of current vehicle (Reference)**, which seems like it would then wake-up the unit in future with _just_ power (and no specific CAN wake-up message).

[image]

For reference; I've also tried sending the **IVI_WakeUpFrame** in dd4all (from **CAN MESSAGE SET C1A_Q4_2017_ALL_MESSAGE_LIST_OFFICIAL_01_12_2017**), to no avail.

For now, I'm left with an unresponsive unit. **If you, or anybody else, is able to provide a candump from an A-IVI-equipped vehicle at startup** (so I can replay it to my unit, and find the wake-up messages), **I'd be _incredibly_ grateful**. 🙌🏼

KarelSvo, 2020-03-15:
If I understand that correctly, IVI sends a wakeup signal with 0644.
I would try using the BCM command to send a wakeup to IVI. 0682

KarelSvo, 2020-10-31:
Have you solved your problem yet?

michaelmcneela, 2020-11-14:
@KarelSvo Nope; gave up. 

Did a ton of research, travelled to France from the UK to get a ZE50 VIN from a showroom Zoe (as ZE50 wasn't here in the UK yet) to look up ZE50 part numbers + parts placement diagrams + part wiring diagrams, compared the internal physical placement & wiring diagrams for the ZE40 Zoe & ZE50 Zoe entertainment systems (and confirmed that the radio unit itself, and the screen, are the same dimensions and seemingly use the same fittings), travelled to the Netherlands to collect a used A-IVI radio unit from a Clio, bought a brand new Zoe-sized A-IVI screen from Renault (£££ 😫), and …

…then couldn't get the radio unit to be detected by DDT4All, and ran out of perseverance.

AlMeu, 2021-06-13:
In DDT4all I can't me too detect it, but with PyREN yes.

@KarelSvo  can you share the dumps what you have for 7" and 9.3"?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif