cedricp/ddt4all #327: CAN line sessions question

primo42, 2020-02-01:
Hi,

what is the purpose of having multiple session modes? Is there a difference if I choose i.e. "Diagnostic - Extended" session (1003) or if I choose "After sales" session (10C0) on CAN Line 1 and then change a setting?

Thanks!

KarelSvo, 2020-02-02:
1003 and 10C0 are different ecu types

primo42, 2020-02-05:
Interesting, I was changing "ESC_15-40-UDS_V1.12 -> ABS/ESC" and "RadNav 2.X v2.5 -> Audio" settings with 10C0 type, although 1003 was the default type for those ECUs. The changes were applied successfully, but I've reversed them afterwards and changed them again with 1003.

shrlnm, 2020-02-05:
1003 - required by standard 
10C0 - vendor (renault) specific. 
Means about the same. Some ECU support only one, some both, others may not require opening session at all for reading parameters (or some of them) 
Actually there are no common rules.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif