cedricp/ddt4all #329: Mirror folding activation on MEGANE III Ph2

kadirayhantr, 2020-02-03:
Hi Everybody, 
I hope I can solve my problem with your big help.
I changed my side mirrors and driver door window switch. I activate mirror folding option on DDT4ALL.  But my mirrors didn't fold.  I couldn't succeed on activating this feature. 
My all sockets and pins are OK and same with an mirror folding featured vehicle.
https://rsmegane.com/threads/retrofit-folding-mirrors.17826/?fbclid=IwAR0fJzZwxK6_WxNs5GbtJ92Xi0ncGa5Ga4y7KOoMT2Ev-05sWMN6OIlRF9c#post-201871   this link succeed of the same project.

I can test my DDCM features with ddt4all like opening windows, moving mirror glasse etc. But when I want to test of mirror folding feature,  dd4all says, '' Bad ELM response : WRONG RESPONSE : NR: SubFunction Not Supported(7F3012) ''   as screenshot. 
[image]

Does any body know what I need to do ?
Does any body have an idea ?

Thanks everybody...

BarektE, 2020-02-03:
Ayna bu bloktan katlanmıyor. Başka bir bloktan katlanıyor, e-mail verirseniz tarfini yaparım.

kadirayhantr, 2020-02-03:
I will be pleasure. Thanks.
kadirayhantr@gmail.com

tanerdogan, 2020-02-08:
Selamlar, ELM cihazınız çin klonu olabilir, daha kaliteli bir cihazla deneyin.

kadirayhantr, 2020-02-08:
> Selamlar, ELM cihazınız çin klonu olabilir, daha kaliteli bir cihazla deneyin.

Selam, Yok dostum sorun modüldenmiş. Modülde aynaların katlanma motoruna giden pin çıkışları yok. Aracın Aynalarının katlanması için DDCM modülünün parça numarası 807310016R  -   807310246R  - 807310596R -  807312028R  numaralarından biri olmak zorunda.

closed 2020-02-08 by kadirayhantr

MouadSoufiani, 2020-12-10:
> Hi Everybody,
> I hope I can solve my problem with your big help.
> I changed my side mirrors and driver door window switch. I activate mirror folding option on DDT4ALL. But my mirrors didn't fold. I [...]

Hi mate, can you send me an email? I want to retrofit those also. I have sended you message on Facebook. Check your spam please

EmreMegane, 2023-07-14:
Sorunu ddt4all uygulaması üzerinden nasıl çözüme ulaştırabilirim ?