cedricp/ddt4all #330: Errors (UnicodeDecodeError: 'ascii' codec can't decode byte / TypeError: 'NoneType' object is not iterable)

vosManz, 2020-02-05:
Hi,

When i try to run ddt4all i get errors, and it can't find any ECU's. I am not sure if this is a bug in the program, or maybe a problem on my computer or hardware.

When i try to run the 'ELM Test' from the first screen, i get an error:
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfc in position 47: ordinal not in range (128)
[image]

But when I connect, the program says it's connected (in the bottom left). I can select my car, and scan for ECU's (with the magnifying glass next to the car-selection) but no ECU's are found. In the console window I see the progress, it takes a couple of minutes, but nothing is found.

When I try to open a screen, the program says it's not connected and keeps trying to reconnect. But in the console window i see for each reconnect there is an error:
TypeError: 'NoneType' object is not iterable
[image]

To check if my ODB tool works, i tested another app to see if there actually is any connection. The app 'ELM327 Identifier' on Android shows the following results:
[image]
A thing i noticed is that when using 'ELM327 Identifier' a couple of lights on the ODB tool are flashing. When I use ddt4all i don't see any flashing lights, only the lights that go on when i put the ODB tool in the port. Is that normal?

I tried the ddt4all tool (Windows version) directly from github (https://github.com/cedricp/ddt4all/releases/download/1.5/ddt4all-installer.exe) but also the ddt4all i found in the issues on this github page (https://www.drive2.ru/b/498093336985338243/, the 'ddt4all All cars.exe', i use the ecu.zip from this package too)

My car is Renault Clio (december 2014)
I use a wifi ODB2 connector (that according to the description should have a PIC18F25K80 in it, but i can't get it open to verify)

Any idea what the problem can be? Is it a problem in ddt4all, my ODB2 device, or maybe my configuration? Any help is appreciated!

KarelSvo, 2020-02-07:
This is clearly an elm with fake answers to the AT commands.
You can use or dispose of it for TorquePro.

vosManz, 2020-02-07:
I was afraid of that, but good to see it verified. Too bad there are so many crappy/fake devices around. By Obdlink sx, do you mean this one?
https://www.amazon.com/ScanTool-425801-OBDLink-USB-Diagnostic/dp/B005ZWM0R4/ref=pd_sim_auto_3?ie=UTF8&refRID=1EMGMTWBYKCR62ACW5PM&tag=scantonet-20

If you (or anyone else) can verify that this one should work with ddt4all i will just buy that one.

KarelSvo, 2020-02-07:
Yes, I mean this interface.
It also includes an original FTDI chip that works with the latest driver from the Windows Update.

vosManz, 2020-02-07:
Thanks KarelSvo for the confirmation. I just ordered that interface so I hope it works well when I receive it and I can start using ddt4all :)
I will close this issue now.

closed 2020-02-07 by vosManz

KarelSvo, 2020-02-22:
> Thanks KarelSvo for the confirmation. I just ordered that interface so I hope it works well when I receive it and I can start using ddt4all :)
> I will close this issue now.
Does ddt4all now work with the new interface?

vosManz, 2020-02-22:
Yes, it does :)