cedricp/ddt4all #331: Voice recognition doesnt work

KEJB1, 2020-02-08:
Hi,
After updating my r-link 2 from 2.2.19.300 to 3.x.x sadly i lost my voice recognition, when i press vr button on stering wheel i got this - There is no database for VR on the current system. Please chek it."
I downgraded my r-link to 2.2.19 but without result. 
My car is Renault Megane 4

Any help would be appreciated

tomidix, 2020-02-08:
After doing the same thing you did I also lost VR only to find out that SPVR is turned on. SPVR stands for Smartphone voice recognition. I enabled Google assistant voice recognition when Android Auto is on and I now use it. It's much more helpful than Renault's.  Anyway if you need Renault try to find SPVR function and disable it in Radnav settings. 
I also hope you are familiar with ddt4all.

KEJB1, 2020-02-08:
I tried to disable/enable SPVR but without result. In both cases r link pops the same notification.

tomidix, 2020-02-08:
Please try this..
With the DDT4all program:

1. Disable the "Embedded VR"optionin RadNav and then restart.
2. Remove the phone associated with the R-link.
3. Reactivate the "Embedded VR"optionin RadNav and then restart.
4. Voice commands should work again.
5. Re-associate the phone to the R-link.
6. When connecting the phone to USB, Android Auto should work correctly.
7. Try briefly pressing the voice command, RLINK2 uses its own system.
8. Try the long press on the voice command, RLINK2 uses the Android Auto .

REFERENCE CAPTURE
[image]

reference tutorial: http://www.foroclub.es/megane4/viewtopic.php?f=420&t=20311

KEJB1, 2020-02-08:
I've tried exactly this method from Spanish forum but without result. 
Any other ideas? 

thanks for your help anyways ;)

tomidix, 2020-02-09:
You should really think about writing and explaining your issue in details. What you have done, by which tutorial.. And what exactly you tried to fix the problem.. Also put the screenshot of the Rlink2 error etc..  Maybe someone recovered your error already but is not aware because you haven't attached picture..

closed 2023-06-02 by Furtif