cedricp/ddt4all #333: How to use ddt4all with a Megane 4 GT

robertalks, 2020-02-12:
Hello, I'm new at this, I got a Megane GT from 2016 and tried to use ddt4all, but when I select the car from the list, there is nothing possible to see or search. I use an USB ELM327, I don't see any errors from ddt4all, like the ELM327 can't read the car or whatever. Any ideas? I also don't understand the need for the ecu.zip, I took the one available in the ddt4all_windows.zip archive, but nothing really happens. Any ideas, or what am doing wrong? Could be that the ELM327 is an older version?

KarelSvo, 2020-02-13:
You are missing the ecu database.
https://www.drive2.ru/b/498093336985338243

robertalks, 2020-02-14:
Cant use that, I'm running on Linux.

robertalks, 2020-02-14:
Nevermind, I was able to 'steal' the ecu.zip from that exe file. Now it works, but I can't enable Android Auto, I mean, it's enabled, but doesn't seem to work. Do I need to do something more or have something more?

tomidix, 2020-02-14:
I think the issue from the subject is solved for you..
Regarding the second question please use translator and read it here:
https://www.foroclub.es/megane4/viewtopic.php?f=420&t=20311