cedricp/ddt4all #334: Activate "Sport Race" on Clio IV Instrument Panel

sirbig, 2020-02-12:
In my Clio IV 1.5 dCi, I can see a hidden message on the instrument panel saying "Sport Race".
I wonder if it can be activated using ddt4all or Pyren...

ECU: Cluster_X98_X87_X82_xHA_X98ph2_X87ph2_xHAph2_v7.9_20181128T110410
[image]

AndyIreland, 2020-02-14:
I'd love to know too

adjimenezv, 2020-04-02:
That only functions on RS Models. You can drive a RS on Normal, Sport or Race Models. I don´t think your DCI hat all the components to activate that.

sirbig, 2020-04-02:
> That only functions on RS Models. You can drive a RS on Normal, Sport or Race Models. I don´t think your DCI hat all the components to activate that.

Well, I know my dCi does not have the components to activate this function, that's why I'm looking for a way to activate it by changing the proper Cluster ECU parameter but it seems that there is no way to do that.
The only solution left would be directly soldering the appropriate components of the Cluster board and that's not worth nor effort IMO.

closed 2023-06-02 by Furtif