cedricp/ddt4all #335: ELM327 Bluetooth  connects but no data

sueandlarry, 2020-02-13:
I am using an ELM327 bluetooth device which the Elm327 identifier app tells me is version 2.2.It works fine with Torque on android devices. Running DDT4all on a PC with bluetooth pairing set up on com 3 it connects and allows selection of the vehicle, ECU etc but returns "no data" to any scan or read operation. Any ideas?

sueandlarry, 2020-02-13:
Update:  I tried the benchmark test with the following result:

"  Using custom DDT database
Trying to open portCOM3 @ 115200
Checking port speed: 38400
Start COM speed : 38400
Traceback (most recent call last):
  File ".\ddt4all.py", line 1201, in check_elm
    res = elm.elm_checker(port, speed, self.logview, core.QCoreApplication)
  File "C:\Program Files (x86)\ddt4all\elm.py", line 1333, in elm_checker
    print cm[2] + " " + res.strip()
  File "C:\Program Files (x86)\ddt4all\Python27\lib\codecs.py", line 369, in write
    data, consumed = self.encode(object, self.errors)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfc in position 47: ordinal not in range(128)"

  I don't know what it means. Does anyone else, and is it related to the "no data" problem?
  Thanks in advance for any help.

KarelSvo, 2020-02-13:
https://github.com/cedricp/ddt4all/issues/330

Marianpol, 2020-02-14:
> I am using an ELM327 bluetooth device which the Elm327 identifier app tells me is version 2.2.It works fine with Torque on android devices. Running DDT4all on a PC with bluetooth pairing set up on [...]

On start up screen try to choose first ELM icon where port speed is 38400.

sueandlarry, 2020-03-13:
Firstly, thanks for your replies and in particular, thanks to Cedric for this wonderful software. KarelSvo was right, it was the same problem as #330, i.e. a phoney scanning device. This is being sold by Spain's foremost car accessories retailer. I have now tried an older usb ELM327 device and it works perfectly.
  I still have one problem however. I am working on a Kangoo 4x4 (2005) 1.9D, and everything works if I select Kangoo I except the ABS which still shows "No Data". There is only one option for an ECU for ABS on the Kangoo I and this does not work. Does anyone have any ideas as to accessing the ABS information on the Kangoo 4x4? The ABS on this model is Bosch and the control unit data lines go to pins 7 and 15 on the diagnostic socket. The protocol with the other working ECU's is not CAN. Thank you in advance for any ideas.

KarelSvo, 2020-03-13:
The Bosch 5.3 file in the Kangoo list is broken. Try Bosch 5.3 from the Megane/Scenic list.

sueandlarry, 2020-03-22:
Thank you KarelSvo for your suggestion, and sorry for the delay in replying. I tried the Bosch 5.3 from the Megane/Scenic but got No Data. I then tried every other xml file for Bosch 5.3 that I could find, but with no luck. I have now connected an oscilloscope to pin 7 of the OBD socket with some interesting results. For all the other ECUs in the car I see constant bursts of data on the K line, presumably in both directions, when the software is set to auto refresh. These ECUs all use KWP2000 protocol and all return data. However the ABS ECU behaves differently. If I am running any Bosch 5.3 xml the software shows ISO8 protocol and the K line remains high with no data in either direction. If however I run any xml such as Bosch 5.7 where the software shows KWP2000 protocol I see regular limited bursts of data which I assume are requests going out, but no data in return. I now suspect a hardware problem in the car, and I am checking whether it is the connections back to the ABS module or the module itself. But why does the software show ISO8 protocol for Bosch 5.3 and send no data? And does the L line, pin 15, play any part in this?
  Hope you can help.

sueandlarry, 2020-03-24:
Update: 
  I should have analyzed the logs earlier: I am getting BUS INIT: ERROR  at the initiation stage (FI or SI) when I try to connect with the ABS ECU irrespective of which xml file I am using, but the other ECUs on the car work fine. I have therefore concluded that there is definitely a hardware fault on the car (wiring or the ABS ECU). I will let you know what I find.
  Thanks to all.
[ecu_ddt.txt](https://github.com/cedricp/ddt4all/files/4377064/ecu_ddt.txt)
[elm_ddt.txt](https://github.com/cedricp/ddt4all/files/4377065/elm_ddt.txt)
[screens.txt](https://github.com/cedricp/ddt4all/files/4377066/screens.txt)

KarelSvo, 2020-03-24:
The only chance for this Ecu is probably DDT2000 and a KKL interface.

cedricp, 2020-03-24:
UCH and ECM are sending frames. It's possible that the ABS ECU is out of order...

KarelSvo, 2020-03-25:
UCH and ECM is KWP Fastinit Monopoint with start command 10C0 and identification 2180.
ABS is ISO 8 without a start command, identification with 51. Maybe the start command by stimulating the L-Line.

closed 2023-06-02 by Furtif