cedricp/ddt4all #339: R-link with the anti theft avtie

Rcarnero73, 2020-02-18:
how to reset a rlink with the anti theft active?

pagrandukas1978, 2020-02-19:
impossible I had that problem

marrius30, 2020-03-11:
Hi, do you have any solutions for this problem?

rttr67, 2021-03-05:
is the solution

Rcarnero73, 2021-03-05:
> is the solution

what is the solution ?
regards

rttr67, 2021-03-06:
hello everything is done on the table intervention inside with the programmer

Koleosguy321, 2021-03-16:
Yeah, sadly no solution as far as I understand.

I've heard people rewrite the VIN using the programmer but for some reason that doesn't work.

You have to virgin mode it before you take it out of the salvage vehicle.

Sucks to say, but I'd call every single dealership you can until you find one that is willing to unlock it (However, a lot of them will refuse for legal reasons). I'd try and get some kind of proof of purchase from the salvage loot to show it wasn't stolen but even though I'd say your chances are slim.

AlMeu, 2021-08-01:
I can reset VIN and atitheft, but if I have the unit on my desk.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif