cedricp/ddt4all #341: GDI - Fuel Rail Pressure

tanerdogan, 2020-02-19:
Hello,

As everyone knows, Renault 1.2 TCe or Nissan Dig-T engine has oil consumption  problem. Renault still refuse this problem and offer to change piston oil segman (without guarantee). But Nissan publish new software update and now less problematic. 

Anyway; Nissan fix problem with change fuel rail pressure because fuel touch chamber and mix with oil and burn together. Also when ppl smell oil via dipstick; its smells fuel too much (fuel&oil mix). Or oil level increase coz fuel leak into crankcase.

So; I can see pressure with DDT4ALL but can not change. I dont think its possible but want to ask, is it possible to change? Or any other software?

https://imgur.com/a/NhDGaez

Thanks.

BarektE, 2020-02-19:
Nobody understands anything here. Here are blind deaf entertainers form

BarektE, 2020-02-19:
Nobody here understands anything. Here, blind people are deaf, they welcome each other

tanerdogan, 2020-02-19:
Hahahah :) Aynen benimde umudum yok ama belki dedim... Aynısını pyren geliştirme sayfasında sordum güzelce cevapladı geliştirici. Velhasıl, CLIP & REPROG lazım lisanslı, onlar da yetkili serviste var, o servislerde pek yetkili değil.

willy53, 2020-02-19:
I am interested in this topic and I will be attentive to its evolution. I take this opportunity to thank this team for their magnificent work. Greetings from Spain

willy53, 2020-04-20:
While waiting for solutions it occurs to me that, at the moment, I am going to use oil with a higher degree of viscosity. Until now I put the 5w40 and I am going to do the test with 10w40, respecting the RN0710 specification of the brand. In a temperate climate where my car circulates there will be no significant difference in cold start and it is possible that it will substantially improve consumption since in a previous car, with a similar problem, it improved significantly.
Cheers

tanerdogan, 2020-04-21:
Hi, thats not effective solution i believe because higher viscosity doesnt good for engine and temporary solution. I investigate the problem and as i said before its about piston oil rings (most of) and happens oil specs and drive habits. You know DI engine has carbon buildup and this carbon can stuck oil rings so what i've done; before oil change engine flush, used 5w30 vis. and new tech SN PLUS oil (especially for LSPI) and fuel system cleaner. 

I did italian tune up for a while and will do every 1000km. Of course its all about wrong engine components and design but oil is important too and Renault offer wrong oil !!! 

I'll follow GM Dexos 1 Gen 2 specs (aka SN PLUS), and dont drive like a grandpa :)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif