cedricp/ddt4all #342: New Captur 2020

cicciofede, 2020-02-20:
Hi everyone! Your job here is really amazing!
I've tried to use DDT program with my Captur 2020. It connects (green on the button part) but when I enter in the ECU it is not able to load any parameters and immediately it diconnects. Is there something I can do? 
I'm using GeekerChip OBD2 WiFi without PIN modification. Thank!

KarelSvo, 2020-02-21:
There is an Elm test program in the ddt4all start menu.
What result do you get?

cicciofede, 2020-02-22:
Thanks a lot for your answer.
Here below:

Apertura porta COM192.168.0.10:35000 @ 115200        (same result with 38400)
Traceback (most recent call last):
  File ".\ddt4all.py", line 1201, in check_elm
    res = elm.elm_checker(port, speed, self.logview, core.QCoreApplication)
  File "C:\Program Files (x86)\ddt4all\elm.py", line 1333, in elm_checker
    print cm[2] + " " + res.strip()
  File "C:\Program Files (x86)\ddt4all\Python27\lib\codecs.py", line 369, in write
    data, consumed = self.encode(object, self.errors)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfc in position 47: ordinal not in range(128)

KarelSvo, 2020-02-22:
The same problem as here https://github.com/cedricp/ddt4all/issues/330

cicciofede, 2020-02-24:
Thanks, I'm waiting a new ELM to check.

cicciofede, 2020-02-24:
I used a different ELM, same log:
 Apertura porta COM192.168.0.10:35000 @ 38400
Traceback (most recent call last):
  File ".\ddt4all.py", line 1201, in check_elm
    res = elm.elm_checker(port, speed, self.logview, core.QCoreApplication)
  File "C:\Program Files (x86)\ddt4all\elm.py", line 1333, in elm_checker
    print cm[2] + " " + res.strip()
  File "C:\Program Files (x86)\ddt4all\Python27\lib\codecs.py", line 369, in write
    data, consumed = self.encode(object, self.errors)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfc in position 47: ordinal not in range(128)

BUT it founds some ECU.

cicciofede, 2020-02-24:
[image]

cicciofede, 2020-02-24:
What about navigation and Cluster/TBD? Need to wait for a sw update?

ulasatar, 2020-05-10:
do you still have problems? I am reading but I cannot write. OBDlink sx

EXcomZkko, 2020-08-02:
Any development about this?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif