cedricp/ddt4all #343: Problem with Read data and write

Saad-BinMansour, 2020-02-23:
Hello
Firstly, thanks for your efforts
Secondly I have a problem and I ask you to solve it: DDT4ALL cannot read or write data and I have tried more than one version of the program and more than one version of the database and the problem remains the same
[image]
[image]

I have attached the Benchmark of the OBD and also the message that comes to me when I change options
[Bunchmark logs.txt](https://github.com/cedricp/ddt4all/files/4242151/Bunchmark.log.txt)
[DDT4ALL logs.txt](https://github.com/cedricp/ddt4all/files/4242153/DDT4ALL.logs.txt)

KarelSvo, 2020-02-24:
Show the file elm_ddt and ecu_ddt.

Saad-BinMansour, 2020-02-24:
this Logs file
[logs.zip](https://github.com/cedricp/ddt4all/files/4247069/logs.zip)

shrlnm, 2020-02-26:
Bad, unprecedented adapter.

CRACKISH, 2020-05-26:
i have same error

[logs.zip](https://github.com/cedric
[image]
p/ddt4all/files/4683447/logs.zip)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif