cedricp/ddt4all #345: Duster II security beeper disable

Daccsika, 2020-02-25:
Dear all,

I would like to disable one function on My Dacia duster II 2018. When I dont turn off the engine and exit the car with keyless card and close either door the car beeps 4 times very loudly. It is annoying me because I always leave the keyless card on My pocket. Anybody can help me with this problem? Thanks a lot!

BarektE, 2020-02-26:
e-posta verin özelden göndereyim

Daccsika, 2020-02-26:
Mazsola.boy@gmail.com

BarektE, 2020-02-26:
sorunu nasıl çözeceğinizi. e-mail ile gönderdim

KarelSvo, 2020-02-26:
> I would like to disable one function on My Dacia duster II 2018. When I dont turn off the engine and exit the car with keyless card and close either door the car beeps 4 times very loudly. It is [...]

https://translate.google.com/translate?hl=en&sl=de&tl=en&u=https%3A%2F%2Fwww.dacianer.de%2Fresources%2Ffunktionen-mit-ddt4all-freischalten-konfigurieren.40%2Fupdates%23resource-update-109

Daccsika, 2020-02-26:
> https://translate.google.com/translate?hl=en&sl=de&tl=en&u=https%3A%2F%2Fwww.dacianer.de%2Fresources%2Ffunktionen-mit-ddt4all-freischalten-konfigurieren.40%2Fupdates%23resource-update-109

It doesn’t help. It is not that beep what i need. :(

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif