cedricp/ddt4all #346: Identification Usm

Guedin88, 2020-03-01:
Bonjour à tous, 
tout d'abord un grand merci à Cédric pour son énorme boulot sur cette application.

Je voudrais ajouter le coming/leaving home sur la télécommande de ma Clio 4, je parviens à me connecter,  je fais une auto identification Usm

[image]
 
Mais ça ne reconnaît pas le calculateur en auto et ça ne correspond à rien dans la liste proposée. 
Pourriez-vous m'éclairer ?

Salutations,  Guedin.

KarelSvo, 2020-03-13:
Maybe this answer will help you https://github.com/cedricp/ddt4all/issues/349#issuecomment-598697008

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif