cedricp/ddt4all #347: leaflet is not visible

emrex89, 2020-03-03:
Hey all, 
The leaflet icon on the dashboard got lost after trying to enable some features. I have a 2017 megane 1.5 dci. It does not have the smart dashboard ( it is conventional needle type)

Could you please help me to retrieve it back to its original?

Thanks!!

BarektE, 2020-03-04:
These programs broke the car of most people, ddt4all, ecu tweaker and PyClip.
In fact, the authors do not know how to use these programs, it is very strange

cedricp, 2020-03-04:
@BarektE What's your theory ? It would be more correct to say that the users broke their car, not the program. Nobody bullied you to connect those programs to your vehicle AFAIK...
Moreover, there's a warning at the application startup saying you can break your car. I first created this application for me to repair my car, then thought it would be a good idea to share it, but it seems I was wrong to do so.
@emrex89, look at the log files in the application directory, there's a file called "screen.txt" containing all parameters at different times. Try to recover original values with the older values. Also, be sure to always work with an XML file that is made for your ECU (use auto ident for that). A good idea is to work on a salvaged ECU first and making screenshots before tweaking.

BarektE, 2020-03-04:
> @BarektE Teoriniz nedir? Kullanıcıların programı değil arabalarını kırdıklarını söylemek daha doğru olur. Kimse bu programları aracınıza AFAIK'a bağlamanız için sizi zorlamadı ...
> Ayrıca, uygulama başlangıcında aracınızı kırabileceğinizi söyleyen bir uyarı var. İlk önce arabamı tamir etmem için bu uygulamayı yarattım, daha sonra paylaşmak için iyi bir fikir olacağını düşündüm, ancak bunu yapmak yanlış görünüyor.
> @ emrex89 , uygulama dizinindeki günlük dosyalarına bakın, farklı zamanlarda tüm parametreleri içeren "screen.txt" adlı bir dosya var. Orijinal değerleri eski değerlerle kurtarmayı deneyin. Ayrıca, her zaman ECU'nuz için yapılmış bir XML dosyasıyla çalıştığınızdan emin olun (bunun için otomatik tanımlamayı kullanın).

At least 5 cars per month are mixed with this program, it comes to me for repair.

Do not be angry, the program is nice, but the language translation can be made in full sentences, not short sentences.

BarektE, 2020-03-04:
Whenever I write a critical article, the person who answers the subject is answered, :))
I like to speak frankly

emrex89, 2020-03-04:
> @BarektE What's your theory ? It would be more correct to say that the users broke their car, not the program. Nobody bullied you to connect those programs to your vehicle AFAIK...
> Moreover, there's [...]

Thank you so much cedric!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif