cedricp/ddt4all #348: R-Link 2 megane ph4 problem with stars in ECO DRIVE2

Smigi44, 2020-03-08:
Hi I do something ?! i have automatic gearbox and somehow after (  don't know ) in ECO DRIVE panel now i have 3 rows star, just like manual gearbox. what can i do ? 

regards and sorry for mistakes, i'm from Poland :)

[image]
[image]

Marianpol, 2020-03-09:
This happend when you done some changes in the R-LINK? Can you share elm_ddt log?

Smigi44, 2020-03-09:
here, just remove .txt from elm_ddt log

[elm_ddt.txt](https://github.com/cedricp/ddt4all/files/4306743/elm_ddt.txt)
[screens.txt](https://github.com/cedricp/ddt4all/files/4306744/screens.txt)

Marianpol, 2020-03-10:
I see R-LINK was updated in 06/02/2020.
"Issue" occurs from that time?

Smigi44, 2020-03-10:
Yes, but not 100% sure

Marianpol, 2020-03-10:
Does it affect your car in any other way than showing your eco driving style in form of stars on the R-LINK screen?

I ran trought your default config, config after update and config after your changes and I don't see any potential options that are responsible for this behavior.
How do you know that R-LINK rate your eco driving like that only in manual gearboxes? Maybe within update, Renault has added that function also for cars with automatic transmissions.

I think it's called an Eco Challenge, and you can turn it off if you want.

Smigi44, 2020-03-10:
Marianpol -NO... and U R right ! i must check it in service toomorow ( I have seller friend) See toomorow here -  I give U anwser.

Smigi44, 2020-03-10:
p.s. Marian - jesteś z Polski ?

Marianpol, 2020-03-10:
> p.s. Marian - jesteś z Polski ?
Tak

Smigi44, 2020-03-10:
Hej, to oficjalnie dziękuje za odpowiedź, jutro sprawdzę to wszystko bo może rzeczywiście masz racje i nie ma problemu ( za duzo renówek w rodzinie). Pozdrawiam serdecznie\

Smigi44, 2020-03-16:
Hej, niestety w innej renówce są dwa rzędy gwiazdek... :( coś jest spieprzone...

Marianpol, 2020-03-16:
Możesz powiedzieć jak został opisany ten dodatkowy rząd gwiazdek? Zdjęcie jest zbyt małe żebym dał radę to odczytać.

Twierdzisz, że te dodatkowe gwiazdki pojawiają się w przypadku manualnej skrzyni biegów. Z Twojej ostatniej konfiguracji mogłem odczytać, że w R-LINK masz ustawioną automatyczną skrzynie biegów czyli tak jak powinno być.

Nigdy nie miałem do czynienia z R-LINK, ale przeglądając opcje konfiguracji oraz Twoje dane nie widzę niczego, co mogło by być za to odpowiedzialne.

Jeżeli jest to powiązane ze skrzynią biegów to można by sprawdzić inne moduły pod kątem tej opcji. Może R-LINK bierze tą informację jeszcze z innego modułu gdzie jest ustawiona manualna skrzynia biegów?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif