cedricp/ddt4all #349: Which ABS/ESC ECU to choose?

AndyIreland, 2020-03-09:
Clio Mk4 16v 1.2 2016 petrol.
When searching for cars ECU's (using magnify glass) it never ever find the ABS/ESC ECU. 

If manually trying to choose there are 2 ECU's showing up for Clio 4 - which correct one to choose?

The 2 Brake ECU's listed are 
ABSESC - X105298 Version 2.6 ABS/ESC

and ABSESC - X10527998 Version 4.0 ABS/ESC

@cedricp  and others

cedricp, 2020-03-09:
Can you copy/paste the log view after a scan ? It's possible that your database is not up to date.

AndyIreland, 2020-03-10:
thanks for replying @cedricp - here are the only ECU's found when doing a search with magnifying glass. The ABS icon light does blink on the cluster at around 37% of doing its scan. but does not show up in the list of found ECU's: the version is 5.5.0 and database is 09/07/2019
[image]

KarelSvo, 2020-03-13:
Just take the latest ABS version.
Renault does not maintain the Autoident data in the ddt2000 database as installed in series production.
As this tool is not relevant for the authorized workshop.
The autoident function would work better, if you use the clip database as an example.
There is identification by Vehicle type and Diag version 
and by some ecus additionally according to Soft.

AndyIreland, 2020-03-14:
> Just take the latest ABS version.
> Renault does not maintain the Autoident data in the ddt2000 database as installed in series production.
> As this tool is not relevant for the authorized workshop.
> The autoident function would work better, if you use the clip database as an example.
> There is identification by Vehicle type and Diag version
> and by some ecus additionally according to Soft.
> 
> ddt4all even declares ecus with a quadruple match as not perfect:
> Found ECU [Cluster] (not perfect match) :Cluster C1A_CMFB_15-40ph2_V2.26_RUN1 DIAGVERSION [22] SUPPLIER [10VisteonPalmela--EV ] SOFT [EV_V06.17] VERSION [1932 instead 1932]

thanks - so its safe to program by manually selecting 

ABSESC - X10527998 Version 4.0 ABS/ESC 

is it?

shrlnm, 2020-03-27:
sometimes, it is possible to find an appropriate xml by the vin number

AndyIreland, 2020-03-28:
> sometimes, it is possible to find an appropriate xml by the vin number

ah right thanks, How do I do that then please?

shrlnm, 2020-03-28:
Give me email or a vin

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif