cedricp/ddt4all #350: Creating ZIP database for android fails

bokkoman, 2020-03-11:
I'm trying to create a zip database for the android version of ddt4all.
I did it before with no problem with an older database. But now I have the latest database and I get a memory error.

```
Writing database
Writing archive
Traceback (most recent call last):
  File "C:\Program Files (x86)\ddt4all\ddt4all.py", line 614, in zipdb
    parameters.zipConvertXML(filename)
  File "C:\Program Files (x86)\ddt4all\parameters.py", line 1713, in zipConvertXML
    f.write(zipoutput.getvalue())
  File "C:\Program Files (x86)\ddt4all\Python27\lib\StringIO.py", line 271, in getvalue
    self.buf += ''.join(self.buflist)
MemoryError
```
What can I do about this?

KarelSvo, 2020-03-13:
Not enough space on your hard drive?

bokkoman, 2020-03-13:
> Not enough space on your hard drive?

I have more than enough space.

cedricp, 2020-03-17:
Think it's more a RAM issue (it uses a lot of RAM to compress)

bokkoman, 2020-03-17:
> Think it's more a RAM issue (it uses a lot of RAM to compress)

Well, that sucks. Maybe you should write a different code to make the zip db? Make it zip constantly instead of at the end, because this way it will be impossible for me to zip this db.

cedricp, 2020-03-17:
Well, I do that on my spare time... I'm actually too busy to work on that, sorry. Try to borrow a computer with more RAM to do the process.

bokkoman, 2020-03-17:
Yeah i know you do that. I'm very happy with the software you provided.
I'll see what I can do with the zipping. Only other laptop I've got is a macbook. But I'm not used to macbooks, and installing ddt4all on mac sounds like a hassle.

naeh, 2020-05-17:
Hello, i have de the same problem here.

`Starting processing ecus\ZIPPERmessagingSimulator.xml 1478/1480 to ZIPPERmessagingSimulator.json
Starting processing ecus\ZIPPER_3_Messaging_Simulator_20150624T100607.xml 1479/1480 to ZIPPER_3_Messaging_Simulator_20150624T100607.json
Writing database
Writing archive
Traceback (most recent call last):
  File "e:\Program Files (x86)\ddt4all\ddt4all.py", line 597, in zipdb
    parameters.zipConvertXML(filename)
  File "e:\Program Files (x86)\ddt4all\parameters.py", line 1544, in zipConvertXML
    f.write(zipoutput.getvalue())
  File "E:\Program Files (x86)\ddt4all\Python27\lib\StringIO.py", line 271, in getvalue
    self.buf += ''.join(self.buflist)
MemoryError`

i was monitoring my ram usage (16go ram, 140go free space on hdd) the ram used didn't grow that much when the bug happen, i think it's a limitation by the system that allows a certain amount of memory to the application (pyton), maybe is there a way to make it allow more memory ?

Et encore merci Cedric pour ce travail de ouf, jamais on te remerciera assez...

Cdt,

jc3dcx, 2020-08-26:
I have the same problem, 

Ram used by process was about a gig, and I have 8gig on laptop with about 100gig space on SSD

[image]

mikegrb, 2021-03-20:
I'm getting: TypeError: Object of type bytes is not JSON serializable
[image]

mikegrb, 2021-03-20:
I've installed from source on Mac and successfully created the zip with the same database so I think this was just an issue in the latest windows installer version.

josefe17, 2023-01-04:
Same issue here as https://github.com/cedricp/ddt4all/issues/350#issuecomment-803212201, even with the same file. Definetely it's an issue with DDT4ALL V2.0 for windows, as V1.5 is compressing fine with the same database (not ended yet)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif