cedricp/ddt4all #351: PROBLEMA RLINK2 KADJAR 2019 (VIRGIN MODE)

Giulio412, 2020-03-11:
Ciao a tutti vi chiedo un aiuto perchè sono disperato..
Per sbaglio ho spinto virgin mode, sono riuscito tramite concessionaria a ripristinare quasi tutto; ho il problema che il navigatore non si avvia proprio, è presente solo l'icona ma come ci spingi su tenta ad aprirsi e torna alla schermata iniziale.
L'altro problema è che il cellulare si collega, ma quando ricevo una telefonata non si sente l'audio.
Riesco a connettermi tramite ddt4all per effettuare modifiche...
Vi prego aiutatemi!
Grazie a tutti

KarelSvo, 2020-03-13:
If you're lucky, you will find your original configuration in your logs/screens.txt file.
Unfortunately ddt4all have no dump function, which means that all configurations can be saved.

Giulio412, 2020-03-13:
Hi, would you explain the procedure to me? thank you very much

shrlnm, 2020-03-15:
@Giulio412 give me your email. Probably I may help you remotely.

Giulio412, 2020-03-15:
giulio.guizzardi85@gmail.com

Thanks so much....

westrafal, 2021-01-31:
I connect to the topic, I did virgin mode unknowingly, I managed to fix it and launched R-LINK, I only have to turn everything on again because half of the functions do not work and it is difficult to set it to the previous state and I am asked if it is possible to install settings from another car to my ?
did not work:
central control panel (navi, multi ..)
360 parking sensors on R-Link
belt assistant
dead pinkt
and many others, I have already repaired a lot but these are not settings for a specific model.
I am asking for help, best regards
tenault Talisman year 2016 1.6 disel

Titiyam14, 2021-08-11:
Bonjour comment avez vous fait pour faire le mode vierge sur ddt4 all merci pour le rlink car je suis avec un écran noir

AlMeu, 2021-08-16:
@Titiyam14 if you have black screen, do you still have sound on speakers?
Do you have flashing the dashboard in two colors?
The USB ports are still working?

Titiyam14, 2021-08-16:
Non je n ai plus de sons,oui j ai le tableau de bord qui clignote en deux
couleurs et les port usb ne fonctionne plus

AlMeu, 2021-08-16:
@Titiyam14 your unit is dead in this case.
You have just to change the unit with another without VIN and antitheft code. With this I might help you, if you like by email (if you share it) in private.

Titiyam14, 2021-08-16:
J ai rdv chez Renault jeudi matin
Autrement il faut quoi pour le remplacer car je connais rien

AlMeu, 2021-08-16:
It's OK also at Renault dealer, but will cost enough well the unit replacing.
They never fix the unit, just replace with a new one.

Titiyam14, 2021-08-16:
Oui je me doute bien ,mais vous pouvez me donner un lien ou autres chose
pour savoir quoi acheter car je n y connais rien et savoir après comment le
configurer merci

Titiyam14, 2021-08-16:
> Yes give me your email and I can give details about what to look.

***@***.***

AlMeu, 2021-08-16:
***@***.***  this I see if you don't log in to write it.

Titiyam14, 2021-08-16:
Tony.cornolti1@gmail.com

maddpower, 2021-10-11:
I'm having the same problem, i have black screen but I have sound and my USB ports are working.
This happened after I pressed the virgin mode.
I am trying to find any solution with no luck.
Anyone can help me?

PetyhO, 2023-05-08:
> @Titiyam14az egysége ebben az esetben halott. Csak ki kell cserélni az egységet egy másikra, VIN és lopásgátló kód nélkül. Ebben segíthetek, ha tetszik e-mailben (ha megosztod) privátban.

Hello [AlMeu](https://github.com/AlMeu)

> Yes give me your email and I can give details about what to look.

Hello.

Please help, my AlMeu email address :)

AlMeu, 2023-05-08:
@PetyhO 
Give me your email and we talk.

PetyhO, 2023-05-08:
petyhotsg@gmail.com

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

hameceq, 2023-06-25:
hello everyone, please i need your help i am really desperate but maybe some one help me. I change my R-link 2 and now it is blocked, I took the original rlink2 safe eeprom and I put on the new one but it is still the same and blocked, can you tell me where to find the R Link ID please I cannot not found. thank you in advance

bonjour toutes les mondes, svp je besoin votre aide je suis vraiment désespéré mais peut-être quelques-un m’aider. je changer mon R-link 2 et maintenant il est bloqué, je pris eeprom sûr rlink2 origine et je mise sur la nouvelle mais toujours pareille il et bloqué, pouvez-vous me dir ou se trouver le ID du R Link svp je n’arrive pas trouvé. merci en avance