cedricp/ddt4all #352: Radio Connect diag unit error

org07, 2020-03-20:
Hi
On the radio, I get the Connect diag unit, error, I ask for help from those who know what's wrong,
[image]
[image]
[image]

org07, 2020-03-26:
Can anyone other than Renault service do this?

The service wants more money from the radio itself.

Does anyone know this job that can help in this forum?

org07, 2020-03-26:
There are a lot of people having this kind of problem, ddt4all will be on the market, and if someone does this, it will help a lot.

Marianpol, 2020-03-27:
Service manual says that your radio configuration is not valid. Check if all options match to your vehicle configuration (check VIN your also).

org07, 2020-03-27:
> Servis el kitabı radyo yapılandırmanızın geçerli olmadığını söylüyor. Tüm seçeneklerin araç yapılandırmanızla eşleşip eşleşmediğini kontrol edin (VIN'inizi de kontrol edin).

Marianpol thank you for your interest, Vin is correct

org07, 2020-03-27:
I think the error is on these, but it won't let me fix it

[image]
[image]

Marianpol, 2020-03-27:
Can you tell me your VIN? Maybe I will can check which option fits to your car.

org07, 2020-03-28:
> Bana VIN'inizi söyleyebilir misiniz? Belki arabanıza hangi seçeneğin uyduğunu kontrol edebilirim.

VF1LZBB0644426153

Marianpol, 2020-03-30:
> 
> VF1LZBB0644426153
I run through radio configuration from your VIN and LC005 configuration NAV0/AMP0/BOX0 is correct, but LC004 configuration should be AUX1/CDC0/SUB0. Try to change it.

org07, 2020-03-30:
> > VF1LZBB0644426153
> 
> VIN ve LC005 yapılandırmanızdan NAV0 / AMP0 / BOX0 yapılandırmanızdan radyo yapılandırması yapıyorum, ancak LC004 yapılandırması AUX1 / CDC0 / SUB0 olmalıdır. Değiştirmeye çalışın.

**thank you i will keep trying**

org07, 2020-03-30:
(00   ) NAV0/AMP0/BOX0
              : (20   ) NAV1/AMP0/BOX0
              : (40   ) NAV0/AMP0/BOX1
              : (60   ) NAV1/AMP0/BOX1
              : (80   ) NAV0/AMP1/BOX0
              : (A0   ) NAV1/AMP1/BOX0
              : (C0   ) NAV0/AMP1/BOX1
              : (E0   ) NAV1/AMP1/BOX1
serviceID    +
              : (63   ) 2E2B1B <Params>
              : (64   ) 1101
              : (65   ) 10C0

##########################  Current values  ##########################

LC005  external elements                                            NAV1/AMP0/BOX0

######################################################################

org07, 2020-03-30:
I cannot change LC005

Marianpol, 2020-03-30:
I forgot to ask - why this DTC showed up? Have you done something to your radio configuration? 

> I cannot change LC005

You cannot change back LC005 to NAV0/AMP0/BOX0? What is showing as a result of this configuration (resp)?

org07, 2020-03-30:
> Sormayı unuttum - bu DTC neden ortaya çıktı? Radyo yapılandırmanız için bir şey yaptınız mı?
> 
> > LC005'i değiştiremiyorum
> 
> LC005'i NAV0 / AMP0 / BOX0 olarak değiştiremez misiniz? Bu yapılandırmanın sonucu olarak ne gösteriliyor?

This is the result **NAV1 / AMP0 / BOX0**

Marianpol, 2020-03-30:
> This is the result **NAV1 / AMP0 / BOX0**

I understand. So LC005 configuration is wrong because you don't have factory navigation system in your car (i.e. Carminat DVD, Carminat TomTom, R-LINK). Change it to NAV0 / AMP0 / BOX0 and if LC004 is not AUX1/CDC0/SUB0 try to change it too.

org07, 2020-03-30:
> > Bu sonuç **NAV1 / AMP0 / BOX0**
> 
> Anlıyorum. LC005 yapılandırması yanlış çünkü arabanızda fabrika navigasyon sistemi yok (yani Carminat DVD, Carminat TomTom, R-LINK). NAV0 / AMP0 / BOX0 olarak değiştirin ve LC004 AUX1 / CDC0 / SUB0 değilse de değiştirmeyi deneyin.

**no navigation, only radio indicator**

org07, 2020-03-30:
[image]

org07, 2020-03-30:
LC005 I changed it, but the radio didn't improve again

Marianpol, 2020-03-30:
> LC005 I changed it, but the radio didn't improve again

Did you change LC004 allso?
I think without answreing my question I will not able to help you.

> Why this DTC showed up? Have you done something to your radio configuration?

org07, 2020-03-30:
> > LC005 Değiştirdim, ancak radyo tekrar gelişmedi
> 
> LC004'ü de değiştirdiniz mi?
> Soruma cevap vermeden sana yardım edemeyeceğimi düşünüyorum.
> 
> > Bu DTC neden ortaya çıktı? Radyo yapılandırmanız için bir şey yaptınız mı?
LC004 yes

org07, 2020-03-30:
I don't know if the numbers in the marked section are correct
[image]

Marianpol, 2020-03-30:
> I don't know if the numbers in the marked section are correct

In default settings they are different, but I don't what are they for and if they are cause of the problem.
So, if you done some changes to your radio please upload a elm_log from software (Pyren or DDT4ALL) you made those changes.

org07, 2020-03-30:
**elm_log** i don't know what it is

Marianpol, 2020-03-30:
> **elm_log** i don't know what it is

For PyRen search for your PyRen home directory then go to pyren/logs. There you should find it.
For DDT4ALL search for your DDT4ALL home directory then go to logs. Thee you find elm_ddt.
Upload both files if you can.

org07, 2020-03-30:
[image]

org07, 2020-03-30:
I'm doing this with Pyren too

org07, 2020-03-30:
TeamViewer can you connect remotely

Marianpol, 2020-03-30:
> TeamViewer can you connect remotely

For now logs will be enough. I need to analyse it and then we can move further.

You have done CF007?

org07, 2020-03-30:
> > TeamViewer uzaktan bağlanabiliyor musunuz?
> 
> Şimdilik günlükler yeterli olacak. Bunu analiz etmem gerekiyor ve sonra daha ileri gidebiliriz.
> 
> CF007 yaptınız mı?

**yes**

org07, 2020-03-30:
**I can send it to you like I uploaded**

[image]

Marianpol, 2020-03-30:
> I can send it to you like

Don't send it for now.

> I uploaded
 
Unfortunately, I don't see it.

org07, 2020-03-30:
> > Sana gönderebilirim
> 
> Şimdilik gönderme.
> 
> > yükledim
> 
> Ne yazık ki görmüyorum.

thank you, I'll be happy when you contact Team Viewer tomorrow

org07, 2020-03-30:
Step by step pyren, operations

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

org07, 2020-03-30:
I'm in the car as of now

org07, 2020-03-30:
the last situation
[image]
### **first part**

[image]

org07, 2020-03-30:
I took a video of the transactions I made, I can send

org07, 2020-03-30:
Video link
[https://youtu.be/pQ01V_iqD_o](url)

org07, 2020-03-31:
> Herhangi bir günlük yüklemediniz çünkü size yardım edemem. Bunu yapın ve belki de yardımcı olabilirim.

I wrote the video link, did you have a chance to watch it?

closed 2020-03-31 by org07