cedricp/ddt4all #354: NO DATA & Bad ELM response: WRONG RESPONSE: Unknown(C)

mbizzaro, 2020-03-28:
Hi Guys, 
Trying to find automatically the module that I can change in my Smart prime 2017. In ddt4all I run the scan functionality but nothing comes up. 
Every time I send data I get: 
Bad ELM response: WRONG RESPONSE: Unknown(C)
I tried looking at all ECU and all of them are giving NO DATA
Is it the connector (https://www.amazon.co.uk/gp/product/B07D3PTX2B/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) ?
What am I doing wrong?

Please find attached:

ELM Benchmark
[EML Benchmarck.txt](https://github.com/cedricp/ddt4all/files/4397495/EML.Benchmarck.txt)

Scan Log
[Scan.txt](https://github.com/cedricp/ddt4all/files/4397496/Scan.txt)

Bad ELM response Log
[Scan.txt](https://github.com/cedricp/ddt4all/files/4397497/Scan.txt)

ddt4all Log
[screens.txt](https://github.com/cedricp/ddt4all/files/4397499/screens.txt)

org07, 2020-03-28:
1. Open the ELM327 See if there is this number on the jeep PIC18F25K80
2.The device on the computer, see the administrator, Is the (COM) part active, a yellow mark.

mbizzaro, 2020-03-28:
HI org07,

Port seems to be fine.
I can't open the ELM327 since it it fused together. On Amazon it is specified that the chip is a PIC18F25K80 (https://www.amazon.co.uk/gp/product/B07D3PTX2B/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1). Is there any way to find out if it's true or not via software?

Thank you

mbizzaro, 2020-04-05:
Hi KarelSvo,

Thank you for your help.
I returned the old terminal. 
Waiting for the new one.
I'll let you know if it had the same issue.

Marco

mbizzaro, 2020-04-29:
> Send ATPPS with this terminal.
> Where is your elm_ddt file?

HI KarelSvo.

Got another ELM327
<http://www.kingbolen.com/elm327-obd2-bluetooth-v1-5-car-diagnostic-tool-elm-327-obd-ii-scanner-chip-pic18f25k80-work-android-ios-windows-12v-diesel>.
Bluetooth this time.
Checked all Bluetooth and com configuration in my Win10 laptop. All good.
The device is working correctly with its own app. Communication and
everything else.
I started the ddt4all and I only get the below

[image: image.png]

It gets stock here.

Below the configuration of the port:
[image: image.png]
[image: image.png]
I checked on the administrative events and I found a warning:

The application-specific permission settings do not grant Local Activation
permission for the COM Server application with CLSID
{2593F8B9-4EAF-457C-B68A-50F6B8EA6B54}
 and APPID
{15C20B67-12E7-4BB6-92BB-7AFF07997402}
 to the user AzureAD\MarcoBizzaro SID
(S-1-12-1-3513732057-1198325939-3368225933-240188653) from address
LocalHost (Using LRPC) running in the application container Unavailable SID
(Unavailable). This security permission can be modified using the Component
Services administrative tool.

There are no log update at all!!

Any suggestion?

Thank you

Marco

KostasAthens, 2021-11-26:
> lHost (Using LRPC) running i

Hello there, have you manage to resolve the NO DATA issue with this new KINGBOLEM ELM? I have just ordered a similar one with the specific chip on it. Thanx, Kostas

AndyIreland, 2021-11-26:
> > lHost (Using LRPC) running i
> 
> Hello there, have you manage to resolve the NO DATA issue with this new KINGBOLEM ELM? I have just ordered a similar one with the specific chip on it. Thanx, Kostas

Hello Kostas - are you choosing the correct ECU to change parameters - did you do a search of ECU's of your car on the CAN network . Sometimes if you choose the wrong ECU you get NO DATA on the red bar

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif