cedricp/ddt4all #355: DDT4ALL for changing "start brand logo" in Smart 453 navigation media system

doctorcico, 2020-03-29:
Hi Friends I'm new member and sorry for my english....I like to "work" with this fantastic tool DDT4all from Cedric on my Smart 453 m.y.2019 with original Touch screen navigation Media System,,,I should change it's original "Smart start logo" with "JBL start logo" or better I should create a new one "Brabus" (see attached picture) but I'm not able to find the right folder where put the new Brabus icon (what is the right size?)
Is it correct the route in ddt4all for change "start logo"? (see attached picture)
Thanks in advance

(Smart 453 and TwingoIII use the same hardware firmware and software)

[image]
[image]

n4th4ndev, 2022-12-31:
I thinks you can't

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif