cedricp/ddt4all #356: Rlink2 v2.2.19.300 to v7.0.24.166 update problem

yk2548, 2020-03-31:
Hello there..

I tried to upgrade the rlink version of my talisman brand vehicle from v2 to v7 myself. the upgrade was successful but there are some problems.

1- When I put the car in reverse, parking sensors and camera are not active and there is no sound
2-eco key and auto park function do not work
3-outside temperature display --- steady in F shape
4-air conditioner temperature values are not visible
5-When I enter and exit the vehicle, the automatic rlink does not turn on or off manually opens but does not close. The rlink closes when the doors are locked.

When I try to downgrade, the old version does not allow to install.

bootloader version: 5446 this has not changed when upgrading to v7.

türkçe versiyon.

merhaba..

talisman marka aracımın rlink versiyonunu v2 den v7 ye kendim yükseltmeye çalıştım. yükseltme başarılı oldu ama bazı problemler var.

1- aracı geri vitese taktığımda park sensörleri ve kamera aktif olmuyor ve ses gelmiyor
2-eco tuşu ve otomatik park işlevi çalışmıyor
3-dış sıcaklık göstergesi ---F şeklinde sabit kaldı
4-klima sıcaklık değerleri gözükmüyor 
5-araca giriş çıkış yaptığımda otomatik rlink açılıp kapanmıyor manuel açılıyor ama kapanmıyor. kapılar kilitlendiğinde rlink kapanıyor.

sürüm düşürmeye çalıştığımda eski sürüm yüklemeye müsad etmiyor.

önyükleyici sürümü: 5446 bu değişmedi v7 ye yükseltirken.

sorry for bad english.

org07, 2020-04-02:
One evening, remove the battery from the pole and wait until morning.
If it doesn't fix it.
Make your settings with ddt4all.

ustunkaya07, 2020-05-24:
Hocam selam. Hangi ildesiniz? Ben de megane 4 için v3.3 ten 7 veya 8 e yükseltmek istiyorum ama yükleme yok diyorlar.

kowalclio2, 2020-06-05:
> Hello there..
> 
> I tried to upgrade the rlink version of my talisman brand vehicle from v2 to v7 myself. the upgrade was successful but there are some problems.
> 
> 1- When I put the car in reverse, parking sensors and camera are not active and there is no sound
> 2-eco key and auto park function do not work
> 3-outside temperature display --- steady in F shape
> 4-air conditioner temperature values are not visible
> 5-When I enter and exit the vehicle, the automatic rlink does not turn on or off manually opens but does not close. The rlink closes when the doors are locked.
> 
> When I try to downgrade, the old version does not allow to install.
> 
> bootloader version: 5446 this has not changed when upgrading to v7.
> 
> türkçe versiyon.
> 
> merhaba..
> 
> talisman marka aracımın rlink versiyonunu v2 den v7 ye kendim yükseltmeye çalıştım. yükseltme başarılı oldu ama bazı problemler var.
> 
> 1- aracı geri vitese taktığımda park sensörleri ve kamera aktif olmuyor ve ses gelmiyor
> 2-eco tuşu ve otomatik park işlevi çalışmıyor
> 3-dış sıcaklık göstergesi ---F şeklinde sabit kaldı
> 4-klima sıcaklık değerleri gözükmüyor
> 5-araca giriş çıkış yaptığımda otomatik rlink açılıp kapanmıyor manuel açılıyor ama kapanmıyor. kapılar kilitlendiğinde rlink kapanıyor.
> 
> sürüm düşürmeye çalıştığımda eski sürüm yüklemeye müsad etmiyor.
> 
> önyükleyici sürümü: 5446 bu değişmedi v7 ye yükseltirken.
> 
> sorry for bad english.
> 
> EDIT:
> 
> I am happy my problem is solved.
> 
> I went to the authorized service for a solution. they upgraded the front loader version from 5446 to 5615 in service. all of my non-working functions have improved.
> 
> Never upgrade from v2 to v7 by yourself, except authorized service. I was lucky my device was not brick but this risk is always there.
> 
> When this problem occurred, I re-encoded with ddt4all, I removed the battery and waited, even I did a hard reset (very bad idea I had to optimize the pages with rdt4all) with ddt4all I did not reach the result.
> 
> the authorized service took care of it within 15 minutes.
> I only paid 63TL.
> 
> Türkçe versiyon
> sorunum çözüldü mutluyum.
> 
> çözüm için yetkili servise gittim. serviste ön yükleyici sürümünü 5446 dan 5615 yükselttiler. çalışmayan bütün fonksiyonlarım düzeldi.
> 
> yetkili servis haricinde kesinlikle v2 den v7 ye kendiniz yükseltme yapmayın. ben şanslıydım cihazım tuğla olmadı ama bu risk her zaman var.
> 
> bu problem oluştuğunda ddt4all ile yeniden kodlama yaptım aküyü söküp bekledim hatta ddt4all ile rlink cihazına hard reset (çok kötü fikir sayfalarca optimizasyon yapmak zorunda kaldım) bile yaptım sonuca ulaşmadım.
> 
> yetkili servis 15 dk içerisinde halletti.
> 63TL ücret ödedim sadece.
> 
> sorry for bad english.

Can you send me files for the V7 update? Any link? My service also tried to upgrade from V 2.2 to V7. During the update, R-Link has stopped working. I was able to bring it back to life with DDT4ALL, but I have the same problems as you. Parking, heating control, Fahrenheit degrees and more. Thank you for notifying me about the need to update the front loader.

RamyarHP, 2021-03-23:
> Hello there..
>
> I tried to upgrade the rlink version of my talisman brand vehicle from v2 to v7 myself. the upgrade was successful but there are some problems.
>
> 1- When I put the car in reverse, [...]

Hello sir, i have the very same problem, i updated mine from v3 to v9 and some of my features dont work, and we dont have renault service in our city, please can you explain how can i fix mine too? thanks

AlMeu, 2021-08-01:
The upgrade from 2.2 or 3.3 to 7.0 is doing only on Renault dealers.
To downgrade maybe is a small chance if passing.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif