cedricp/ddt4all #357: ddt4all android hangs choosing usb without it

pakoto, 2020-03-31:
When you select usb unintentionally, app close with error when start. You need to reinstall or try to switch quickly to wifi or bt before crash.

Roms33, 2020-04-25:
Same problem

pakoto, 2020-04-28:
more people could recreate bug? 

If you have elm bluetooth or nothing, select usb or wifi into ecu tweaker, and try to open app again. If it closes after 1 sec, please report and put android version here. You can try to open and quickly change to bluetooth and works again. Reinstall also works.

Roms33, 2020-04-28:
I use elm usb. The app close when i choose usb. Android 10.

localhost61, 2020-11-08:
No matter the USB device, you don't need to connect it to get the force close (French Android 9 for me)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

MTPro94, 2024-05-09:
On Android 13 Usb Elm doesn't work, it doesn't manage to find Ecu in Megane III 2014, or it closes when I select USB. I installed the application version 1.8, 1.85, 1.9. I copied ecu.zip to the internal memory, I think I did the steps correctly. Does anyone know how to solve it? Thank you .