cedricp/ddt4all #358: Probleme avecOBDLinks SX

zavzb, 2020-04-01:
Bonjour,

J'ai voulu essayer ce logiciel qui me semble prometteur, et en plus open source! Je travaille sous W7 32b, avec une interface OBDLink SX USB.

Lorsque je tente de me mettre en mode connecté, j'ai le message d'erreur suivant.
[image]
Je ne comprends pas pourquoi windows me refuse l'acces à ce port.
J'ai essayé differente configuration; changer le port com, changer la vitesse du port, etc.... rien n'y fait.
Je tiens à préciser que j'ai cette interface à la base pour fonctionner avec renolink, et qu'elle fonctionne super bien dessus.

J'ai lu dans un autre topic que cela était dû à la pietre qualité du clone, mais le mien est un original si je ne dis pas de bêtise. Je suis en firmware 4.5.1.

J'espère avoir donné suffisament de renseignement.
En tout cas merci encore à cedric, tu es formidable.

KarelSvo, 2020-04-03:
http://disc.elm327rus.ru/software/els27/ELS27Baudrate.zip
Renolink uses a higher baud rate than 115200.
With this tool you can change.

zavzb, 2020-04-04:
Thks for youre answer.
I tried with you soft, but can,t change baudrate, so I tried with Ifprepare2 including in Renolink, when I changed it, disconnect and reconnect the interface, and it works.

Thanks very much, I hope this topic can be usefull.

skyloker, 2020-10-15:
Bonjour, j'ai le mème problème. je ne peux plus me connecter avec dd4all. Après connexion du logiciel RenoLink. Pouvez vous  expliquer plus précisément comment vous avez solutionner le problème ? C'est un logiciel Ifprepare2 ?
Merci d'avance

zavzb, 2020-10-15:
Bonjour, oui j'utilise le logiciel fourni avec renolink pr modifier le baudrate !

KarelSvo, 2020-10-15:
> Bonjour, j'ai le mème problème. je ne peux plus me connecter avec dd4all. Après connexion du logiciel RenoLink. Pouvez vous expliquer plus précisément comment vous avez solutionner le problème ? C'est un logiciel Ifprepare2 ?
> Merci d'avance
this solution is more convenient
https://www.tlemcen-electronic.com/forum/showthread.php?p=813479#post813479

skyloker, 2020-10-15:
Bonjour,
Il faut être inscrit sur le site pour pouvoir téléchargé ce fichier "renolink.ddt4all.zip"
Enfin si il bien de celui ci.. J'ai parcourus tous le lien et il est proposez ce fichier

Marianpol, 2020-10-16:
I think the best way to change OBDLink baudrate is to download STN Term from here:
https://www.scantool.net/scantool/downloads/utilities/

Run this program.
Choose OBDLink COM Port
Choose actual baudrate (I believe it is propably 500000 or 1000000) You can check if you selected a correct baudrate by simply sending a ATI command (you can use button to send that command) and verifying if received message is not distorted.
If is not, you found a actual OBDLink baudrate.
Now, you will have to change a baudrate to baudrate used by DDT4ALL e.g 115200.
To do so, you need to use STSBR command with desired baudrate value.
In this example command will be STSBR 115200
Write a command in a box in the STN Term, send it and you should get an OK info.
Now, you need to switch Baud rate settings in the STN Term to one, that you have previously set.
Send again ATI command to see, if you can communicate with OBDLink correctly. 
If so, write a STWBR command and send it to save selected baudrate. 

Without sending the STWBR command your selected baudrate will not be saved and it will fallback to previous value after OBDLink unplug or connecting to DDT4ALL (DDT4ALL sends a ATZ command with resets OBDLink).

Your OBDLink baudrate has been changed and you can connect to a DDT4ALL again.

KarelSvo, 2020-10-17:
> Your OBDLink baudrate has been changed and you can connect to a DDT4ALL again.

It's about the baud rate setting of the FTDI level converter in the OBDLINK interface.
https://www.ftdichip.com/Support/Knowledgebase/index.html?ft_setbaudrate.htm
In the linked solution above, the baud rate is not changed by the level converter of the interface.
But added in ddt4all, which can be used with 1000000 baud settings from renolink.

fellouri79, 2021-02-28:
https://www.youtube.com/watch?v=ocegmhEKhWA

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif