cedricp/ddt4all #365: Symbol 2014 hız sabitleme

volkann60, 2020-04-15:
Merhaba öncelikle Google translate kullanıyorum dilinizi bilmiyorum. 

Renault Symbol 2014 model aracım var. Hız sabitleme ve limitleme özelliğini açmak istiyorum ancak hata alıyorum konu ile ilgili yardımcı olabilir misiniz.

Aldığım hata

Sedding elm request: 2E222F01

Bad elm response: wrong response: rn: comditions not correct or request squence error (7F2E22)

volkann60, 2020-04-15:
Symbol 2014 cruise control

Hello, first of all, I use Google Translate, I don't know your language.

I have a Renault Symbol 2014 model vehicle. I want to turn it on for speed stabilization and limitation, but I am getting an error. Can you help with the issue?

The error I got

Sedding elm request: 2E222F01

Bad elm response: wrong response: rn: comditions not correct or request squence error (7F2E22)

volkann60, 2020-04-15:
[image]
[image]
[image]
[image]

volkann60, 2020-04-15:
[image]
[image]
[image]

volkann60, 2020-04-15:
[image]

BarektE, 2020-04-16:
tesisatı hazır var mıydı yoksa kendiniz tesisat mı çektiniz? Direksiyon simidi üzerindeki 331 kodlu düğmelerin direnç değerlerini okuyamıyor enjeksiyon beyni

closed 2020-04-20 by volkann60