cedricp/ddt4all #369: About solving error code (261)  Megane 3 X95 Multimedia problem.

selmansanli, 2020-04-17:
i have aproblem with my megane 3 Multimeda : i can not use radio/cd/bluettooth and all of the fuses are right...  CAN CLIP device not see MULTIMEDIA. I give 261 ERROR CODE.  I have ordered  OBD2 with orginal pic185f25k80. Can i solve this problem with ddt4all ? I think if i renew configuration problem will be solve..

Sorry for my bad english...

org07, 2020-04-19:
Ben uzun süre uğraştım yaptırmak için sağ olsunlar buradan bir arkadaş ve PyRen yazılımcısı ShrLnm de yardımcı oldular ama çözüm olmadı. İstanbul'da Renault oto elektronikçisi var o usta çözdü. Hata olarak beyin tanıtılmamış hatası veriyordu. 3 tane servisi gittim birisi 350 TL + KDV dedi diğer ikisi nasıl yapılacağı hakkında bilgileri de yoktu.

selmansanli, 2020-04-19:
hocam adınız nedir? sizinle telefonda görüşebilir miyim?

selmansanli, 2020-04-19:
is anyone can help about  Megane 3 sleep multimedia  ???  after 2 days  i give  ddt4all  logs. 
 When i connect the Can clip to the car i give **error code 261** = multimedia not found!
  
Can clip  not see the multimedia system.  Some people say me this problem will be solve by " change configuration file"

my multimedia system has tomtom gps navi. when i  push "ON" button tomtom screen is wake up!
and i see the map on screen! BUT radio/cd/bluetooth/AUX never give me sound.

if anyone will help me i can add video and photos ...

look below what is mean of 34HY and 34HU.   

how can i use these codes with ddt4all.

Radio (261)
-- | -- | -- | --
Connector | Pin | Code | Description
A | 1 | 34A | Left Hand Rear Loudspeaker + Signal
A | 2 | 34G | Left Hand Front Loudspeaker + Signal
A | 3 | 34E | Right Hand Front Loudspeaker + Signal
A | 5 | BP1H |  
A | 6 | 34B | Left Hand Rear Loudspeaker - Signal
A | 7 | 34H | Left Hand Front Loudspeaker - Signal
A | 8 | 34F | Right Hand Front Loudspeaker - Signal
A | 9 | 34D | Right Hand Rear Loudspeaker + Signal
A | 10 | 34C | Right Hand Rear Loudspeaker - Signal
A | 12 | MT | Radio Electrical Earth
B | 1 | TB39 | Am/Fm1 Radio Shielding Braid
B | 2 | 34EP | Am/Fm1 Radio Signal
**C | 1 | 34HY | Multimedia Stop/Start Control
C | 2 | 34HU | Multimedia Wake-Up Signal**
C | 7 | 107F | Multimedia Microphone Signal
C | 8 | 107G | Hands Free Telephone Microphone +
C | 19 | 107W | Diagnostic Socket Can H Multimedia Signal
C | 20 | 107X | Diagnostic Socket Can L Multimedia Signal
C | 23 | 107J | Telephone Microphone Screening

org07, 2020-04-19:
> hocam adınız nedir? sizinle görüşme miyim?

Benim radyo bu https://youtu.be/tzE1GsITl_w

selmansanli, 2020-04-23:
CAN CLIP NOT SOLVE THIS PROBLEM then i was make a decision to look ICs in my radio ...  because can clip can not see the multimedia(radio) device.  When i open the radio device see TDA7564B  sound out IC.  this ic can comminicate with canbus already. You can see below alldatasheet document.
Another ic is TJA1040. TJA1040 is COMPLETELY  work with canbus. The TJA1040 is the interface between the Controller Area Network (CAN) protocol controller and the physical bus
I give you information after test these ic's...

Im waiting your advices..

[image]
[image]
[image]
[TJA1040.pdf](https://github.com/cedricp/ddt4all/files/4523193/TJA1040.pdf)
[datasheet.pdf](https://github.com/cedricp/ddt4all/files/4523211/datasheet.pdf)

shrlnm, 2020-04-24:
Have you checked the OBD pinout?

According to your scheme 

C | 19 | 107W | Diagnostic Socket Can H Multimedia Signal
should be connected to OBD pin 13

C | 20 | 107X | Diagnostic Socket Can L Multimedia Signal
should be connected to OBD pin 12

selmansanli, 2020-04-24:
> Have you checked the OBD pinout?
> 
> According to your scheme
> 
> C | 19 | 107W | Diagnostic Socket Can H Multimedia Signal
> should be connected to OBD pin 13
> 
> C | 20 | 107X | Diagnostic Socket Can L Multimedia Signal
> should be connected to OBD pin 12

These pins (12-13) for multimedia canbus connect. There is not problem with this pins. I was buyed elm327. If it is ok ( genunie 18f25k80 chip ) i will show you ddt4all log.

selmansanli, 2020-04-25:
> OBD pin çıkışını kontrol ettiniz mi?
> 
> Şemanıza göre
> 
> C | 19 | 107W | Diyagnostik Soket Can H Multimedya Sinyali
> OBD pin 13'e bağlanmalıdır
> 
> C | 20 | 107X | Diyagnostik Soket Can L Multimedya Sinyali
> OBD pin 12'ye bağlanmalıdır

Dear shrlnm,
I get a new ELm327(bluetooth) below some photos from my android.  when i try to coonect AUDIO UNIT i get these messages look below, 
[image]
[image]
[image]
[image]
[image]
[image]
[VISU SCHEMA_RADIO.PDF](https://github.com/cedricp/ddt4all/files/4532741/VISU.SCHEMA_RADIO.PDF) Look that document Code 261 is about the radio. Radio is ok when i push power botton navi screen comes but i can't get any sound.

shrlnm, 2020-04-25:
Hello @selmansanli Have you modified the elm for working with multimedia CAN bus? The radio is connected to pin 12 and 13.

selmansanli, 2020-04-25:
No I don't modify elm. I get it yet.  When i change the 12-13 connections, i put the new datas here. There is a curfew these days its about CORONA virus. i need 2 position switch for make connection.

selmansanli, 2020-10-01:
Now,i have modified my elm device for give audio fault codes.  12-13 pin are activated for get multimedia data. 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

@shrlnm 

My car is 2010 year megane 3 Phase1.  Tomtom navi is ok ! But can not get sound from: **radio/cd/bluetooth/Aux**

i need default adjustments can u help me ?

i don't know how to get log files from ddt4all.

selmansanli, 2020-10-01:
@shrlnm  similar problem >>>
https://www.club-renault.ru/forum/topic/5528-diagnostika-svoimi-rukami/page/202/

shrlnm, 2020-10-02:
@selmansanli give me your e-mail

selmansanli, 2020-10-02:
mail: selmansanli@hotmail.com

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 good tip share it as discutions

closed 2023-06-02 by Furtif