cedricp/ddt4all #370: Create new Screens

PetarKleut, 2020-04-18:
Hi,
I would like to create some new screens. In edition mode I can create new category and screen and after pressing UI edit button I can create new label and button but that's all. 
How do I add parameters to the screen?

I do get an error "QLayout: Cannot add null widget to QHBoxLayout/" maybe that's disabling some part of UI. 

Thanks

PetarKleut, 2020-04-21:
I was able to create the screen manually. After going to File/Save current ECU the file with the extension json.layout will be saved in json folder. Then I was editing this json.layout file using one of the ECU's from the database that I had as template. 
Quite slow but maybe it helps somebody with same issue.

cedricp, 2020-04-21:
Oh, yes, haven't had time to finish that feature. But it should work this way, though

KarelSvo, 2020-05-10:
> Hi,
> I would like to create some new screens.

I use this easier way:
Edit the xml files with the ddt2000 editor and only then create an ecu.zip archive.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif