cedricp/ddt4all #372: Connection to ELM was Lost. Port Closed

nerdinary, 2020-04-22:
I'm using a wifi obd2 adaptor - seems to work fine with android apps. I can connect to it using ddt4all and things seem to work fine, till I select an actual ECU and then the program hangs.

On the console, I get a flood of messages saying:

"Connection to ELM was lost. Port Closed".

Only way to recover is to kill the program and try again.

nikoul, 2021-06-07:
I have the exact same issue. @nerdinary did you find a solution eventually ?

11noobestplayer, 2021-06-09:
hi
same problem for me to, any solutions pls ?

nerdinary, 2021-06-11:
Unfortunately I never found a solution. I've given up.

KarelSvo, 2021-06-12:
Far too little information to analyze the problem remotely.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif