cedricp/ddt4all #373: Keyless Vehicle

AndyIreland, 2020-04-23:
hello all , anyone got an Idea about why when I go into EMM Ignition that it says  Cbx_keylessVehicle_CF that it is set to '0' on my car (Clio 4 1.2 16v 2016) when the car has no physical key but a KeyCard instead? 
was tempting to see what would happen if I changed value to '1' but a bit cautious  just in case i break something?

Pic:
[image]

org07, 2020-04-24:
Cbx Keyless Vehicle not change from 0 to 1 ddt4all can not

org07, 2020-04-24:
Cbx Keyless Tool does not change from 0 to 1, ddt4all cannot do this

AndyIreland, 2020-04-25:
> Cbx Keyless Tool does not change from 0 to 1, ddt4all cannot do this

Oh, right so if i go to change from 0 to 1 it just wont accept it and stay on '0' anyway?

org07, 2020-04-26:
Will stay at 0

org07, 2020-04-26:
Will stay at 0 will not go to 1

AndyIreland, 2020-04-26:
oh right - thank you. and thanks for the reply

MohammedTarek14, 2020-06-02:
> Cbx Keyless Tool does not change from 0 to 1, ddt4all cannot do this

Which program can do it?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif