cedricp/ddt4all #376: Recovering acoustic parameters from logs?

dwilliams782, 2020-04-25:
Hi All,

Whilst trying to fix my sat nav, it appears I've lost my acoustic parameters. I have a feeling I reset them whilst poking around but I'm not 100% sure.

Is there any way I can recover these from local log files? Or regenerate them? I'm missing the following:

acoustic_driver
acoustic_whole_car
phone_acoustic_parameter

Thanks in advance..

shrlnm, 2020-04-25:
Hi
If you do not have a log then give me you e-mail

dwilliams782, 2020-04-25:
Hi shrlnm,

I looked through the logs but the only entries for the audio ECU show null responses for the acoustic parameters so it's likely they were wiped already (I bought the car with the TomTom faulty).

You can reach me on williams.d.1242@gmail.com. I'll edit the email out of this comment after.

selmansanli, 2020-04-25:
one of log:

[24-04-04:40:39]  Log file created
[24-04-04:54:39]  Log file created
[24-04-04:54:43]  Log file created
[24-04-05:21:50]  Log file created
[24-04-05:25:20]  Log file created
CAN SENT: [24-04-05:29:19] 10C0
CAN RECV: [24-04-05:29:19] ERROR : 0210C0
UNABLE TO CONNECT
CAN SENT: [24-04-05:29:19] 2180
CAN RECV: [24-04-05:29:19] ERROR : 022180
UNABLE TO CONNECT
CAN SENT: [24-04-05:29:20] 1003
CAN RECV: [24-04-05:29:20] ERROR : 021003
UNABLE TO CONNECT
CAN SENT: [24-04-05:29:21] 22F1A0
CAN RECV: [24-04-05:29:21] ERROR : 0322F1A0
UNABLE TO CONNECT
CAN SENT: [24-04-05:29:21] 22F18A
CAN RECV: [24-04-05:29:21] ERROR : 0322F18A
UNABLE TO CONNECT
CAN SENT: [24-04-05:29:22] 22F194
CAN RECV: [24-04-05:29:22] ERROR : 0322F194
UNABLE TO CONNECT
CAN SENT: [24-04-05:29:22] 22F195
CAN RECV: [24-04-05:29:22] ERROR : 0322F195
UNABLE TO CONNECT
New session with ECU CBox V0.6
[24-04-05:42:02]  Log file created

dwilliams782, 2020-04-25:
Does the screens.txt get overriden each time? That log file only has entries from today whereas my elm file has everything from yesterday when I started playing with this?

cedricp, 2020-04-26:
No, new logs are appended

dwilliams782, 2020-04-27:
[image]

shrlnm was able to supply me with a full radio configuration, however I no longer believe that my parameters are wiped. In the photo above, it appears that I am getting sensible data back, but ddt4all / the ecu file / my adaptor is unhappy with it, hence "NO DATA" appearing.

I've tried this with flow control enabled, various baudrates and delays / timeouts set, no change. Is it possible that my chinese adaptor isn't up to the job?

[benchmark.txt](https://github.com/cedricp/ddt4all/files/4541438/benchmark.txt)

I've also attached the output of my ELM benchmark, there are a number of failures in there, is any of this linked?

Thank you for an excellent tool btw.

cedricp, 2020-04-27:
Frames are very long, maybe the ELM memory is overflowed... Would be interesting to test with a better/faster adapter like an OBDLink.

dwilliams782, 2020-04-27:
Do the more expensive ones (such as obdlink) still need to be opened and modified to support the media can bus?

cedricp, 2020-04-28:
It should indeed, but never tried with so long frames on my side...

pakoto, 2020-04-28:
i confirm that fixed some medianav with obdlink sx, sending acoustic parameters ok

dwilliams782, 2020-04-28:
> It seem that "Flow Control" was not actually enabled. With flowcontrol enabled ecu transfer to elm only defined number of frame at once to prevent buffer exhaustion.

> It should indeed, but never tried with so long frames on my side...

Should this apply to reading as well? I do not expect the error that I get in the screenshot.

> i confirm that fixed some medianav with obdlink sx, sending acoustic parameters ok

Can you confirm how you modified the obdlink sx? My cheap elm327 adaptor came with a switch; did you have to build something external?

pakoto, 2020-04-28:
I do it with medianav "on table". Connecting medianav with 12v and canbus L H at home.

I also have an adapter to use dacia clio iv clusters at home:

https://www.youtube.com/watch?v=8vJiZ3MinfA

Nothing special here: 12v adapter, OBD plug and renault plug from junkyard.

shrlnm, 2020-04-28:
> i confirm that fixed some medianav with obdlink sx, sending acoustic parameters ok

Do you have a log of writing acoustic parameters with obdlink sx? It really interesting.  I saw a lot of logs with unsuccessful tries made with ordinar ELM. The result was always with NR serviceNotSupportedInActiveSession. It is looks like the session times out occur while we writing long command.

shrlnm, 2020-04-29:
@dwilliams782 I've checked the logs you sent me once again and confirm that the flow control was not enabled actually. When it is enable the elm_log file should contains a pattern 'CFC0'

dwilliams782, 2020-05-02:
@shrlnm that's interesting - it was enabled so I don't know what happened there.

I received an OBDLink SX today, and modified it for the MM bus. With this adapter (and the delay set to 1000ms) the acoustic parameters loaded correctly, so as I had started to suspect, they were never actually wiped in the first place.

That left me with the error message "please connect diag" still for the radio, but all configuration appearing correct. I eventually got this error to disappear by disabling the front tweeters (not sure why this made any difference).

The radio then spent two minutes beeping loudly, then prompted for the radio code. This appears to be some sort of retuning procedure.

I'm now in the position where the TomTom shows the radio and I no longer get the "connect diag" error, however all I get is hissing from the front left speaker. I get no other audio whatsoever.

I will continue trying to fix this tomorrow.

shrlnm, 2020-05-03:
@dwilliams782 Thank you for information. It is interesting. Please inform us about your experiments.

shrlnm, 2020-05-03:
@dwilliams782 I just looked into CLIP database and found that there are speaker tests and configuration scenario for your radio module

`
ECU : 11338  FG7610728.XML

1  - DE : Device errors
2  - ID : Identifications
3  - SY : System state
4  - SP : System parameters
5  - LC : System configuration
6  - AC : Executing tests
7  - CF : Changing configuration
8  - VP : VIN programming
9  - RZ : Resets
10 - ED : DE extra information
11 - FAV : Favourite Parameters
12 - DDT : DDT screens
Q  - Up
Choose :

`

`
ECU : 11338  FG7610728.XML
Screen : AC : Executing tests -> MULTIMEDIA -> AUDIO
1  - AC004 [Command] FRONT RIGHT-HAND SPEAKER LINE TEST
2  - AC005 [Command] FRONT LEFT-HAND SPEAKER LINE TEST
3  - AC006 [Command] REAR RIGHT-HAND SPEAKER LINE TEST
4  - AC007 [Command] REAR LEFT-HAND SPEAKER LINE TEST
5  - AC010 [Command] MICROPHONE TEST
Q  - Up
Choose :
`

`
ECU : 11338  FG7610728.XML
Screen : CF : Changing configuration -> C007 [Command] 

#############################  Command parameters  #############################

Prerequisite : If the vehicle is equipped with a CAN gateway, before activating the command, please check that the CAN gateway is in Open mode via the "Gateway security level" menu.
Ignition on and engine stopped. For vehicles fitted with automatic transmissions, position the lever in P or N. For electric vehicles, unplug the vehicle charging lead. When the command is finished, return to the FAULTS screen before switching off the ignition.

name         : C007
codeMR       : CF007
label        : VEHICLE CONFIGURATION
type         : CF
scenario     : 
inputlist    +
serviceID    +
              : (46   ) 2E2B2B
              : (47   ) 2E2B2C
              : (48   ) 2E2B2D00011E1422241A14280D0A081F0F020206020408300301010001001430121220121400800100640A050F011402080A043C0F0F140F321E3250050F0009120A8CE10A5A5A0000080100003C961401C80000050000000000000000
              : (49   ) 2E2B43AF70503C32281D0F070501
              : (50   ) 2E2B44FAAF70523C2D1E0F090706
              : (51   ) 2E2B25
              : (52   ) 2E2B01010708
              : (53   ) 2E2B0809
              : (54   ) 2E2B1201
              : (55   ) 2E2B0D00
              : (56   ) 2E2B1100
              : (57   ) 2E2B1308
              : (58   ) 2E2B1B60
              : (59   ) 2E2B1C20
              : (60   ) 2E2B2400
              : (61   ) 2E2B26010109
              : (62   ) 1101
Are you ready to execute the command? <yes/no>:

`

dwilliams782, 2020-05-03:
SUCCESS!!!

A massive, massive thank you to @shrlnm. The hissing / static noise was down to incorrect acoustic parameters after all - using the configuration you sent me, I was able to write the acoustic parameters using DDT4all and my OBDLink SX and it burst into life.

I would not have been able to recover this without your help, and this excellent tool. I will do a write up later on the RS megane site to share my findings.

closed 2020-05-03 by dwilliams782

hdano1, 2021-02-14:
> [image]
> 
> shrlnm was able to supply me with a full radio configuration, however I no longer believe that my parameters are wiped. In the photo above, it appears that I am getting sensible data back, but ddt4all / the ecu file / my adaptor is unhappy with it, hence "NO DATA" appearing. BTW this is the "correct" ecu file that autoscan finds (although I have my doubts).
> 
> I've tried this with flow control enabled, various baudrates and delays / timeouts set, no change. Is it possible that my chinese adaptor isn't up to the job?
> 
> [benchmark.txt](https://github.com/cedricp/ddt4all/files/4541438/benchmark.txt)
> 
> I've also attached the output of my ELM benchmark, there are a number of failures in there, is any of this linked?
> 
> Thank you for an excellent tool btw.

Can you please share those full radio config ?
Same thing happened to me. I accidentaly did a hard reset of radio, now I have message "connect diag "

leader21, 2022-06-26:
sorry to dig out this very old issue, but i have an arkamys radio here with a "connect diag" error too and another arkamys radio with a totally flat sound, almost like listening to music in a tin can ;) the radio is configured as R2_V3.81 and my Navigation screen is a NFAv0.6.

so, can you provide me with the radio config too? I must get rid of that awful sound.

i tried to search so many forums and other websites to get useful information about the sound issue but got no useable results.
i have another radio here, no arkamys but it is shown as R2_V3.81 too. Can i take the acoustic parameters from that radio and just "copy/paste" it? the sound of that radio is good, but it is no arkamys and has an old software version installed. 

maybe anybody else has working acoustic parameters vor me? 

the radio with the sound issue has the part no. 281157550R
the radio with the connect diag problem has the part no. 281159243R

jasonpsa, 2025-05-20:
I have the same problem. The radio write "connect diag unit" after the mechanic change the turbo and clear the fault with the diagnostic unit. I have the ddt4all and the elm328 with pic. I have played with the radio configuration without success. Could someone tell me what I need to do?  Thankss