cedricp/ddt4all #377: Successfully rewired CAN adaptor but cannot find radio when I do a search

AndyIreland, 2020-04-29:
Clio Mk4 2016 Medianav Evolution 9.1.3 ULC 2.2v
Hello all I got a ODB Lead today and successfully shorted out pins pin 6 to pin 13  and pin 14 to pin 12   and done a search using magnifying glass in DDT4ALL  but it did not find the ULC Radio. 

So I manually selected the nearest I could get was ULC _Gen2 V1.5  (I could not find ULC 2.2v) - in a any case the parameters read OK .. if I change any of the parameters in ULC V1.5  will it break my radio if I change the settings for ULC 1.5  even though my radio is ULC 2.2v ?

Thanks.

BarektE, 2020-05-01:
Arama yaptırdığınızda 2 tane bulacak biricisi (MIU bu ekran) ikincisi radyo ama radyoyu bazen göstermiyor radyo üzerinde işlem yapmaya olanak sağlıyor. Radyoyu siz kendiniz manüel olarak seçeceksiniz. Kolay gelsin

BarektE, 2020-05-01:
tavsiyem şudur ddt4all ile kesinlikle radyo bölümleriyle oynama çabuk kırılıyor.

KarelSvo, 2020-05-10:
> 
> So I manually selected the nearest I could get was ULC _Gen2 V1.5 (I could not find ULC 2.2v) - in a any case the parameters read OK .. if I change any of the parameters in ULC V1.5 will it break my radio if I change the settings for ULC 1.5 even though my radio is ULC 2.2v ?

No problem, if you don't touch for the Configuration Reset button.

AndyIreland, 2020-05-12:
> No problem, if you don't touch for the Configuration Reset button.
thank you @KarelSvo

BarektE, 2020-07-10:
AndyIreland hello. clio 4 mirror folding feature in which block

BarektE, 2020-07-10:
Merhaba AndyIreland, Clio 4 mirror folding feature in which block.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif