cedricp/ddt4all #378: compatibilité DDT4ALL

agn22, 2020-05-07:
Bonjour,
J'ai une interface EML327 V1.5 avec une puce PL-2303HX
J'ai installé DDT4ALL pour activer android auto sur ma mégane3 mais le bouton mode expert reste grisé, je n'ai donc pas accès aux menus déroulant
Quelqu’un saurait il si cette puce est compatible ?
J'ai cherché sur le forum mais je ne trouve pas la réponse.
Merci d'avance :)

cedricp, 2020-05-07:
Le bouton expert ne sert qu'a bloquer les requetes d'ecritures par les boutons.
La puce PL2303 est le convertisseur USB vers bus serie, il faut verifier le chip ELM qui doit etre du type PIC18F...
Quelle version utilises tu ?

agn22, 2020-05-07:
Merci pour ta réponse rapide, en effet, je n'avais pas regardé du bon côté de la carte électronique.... c'est bien un PIC18F25K80.
pour la version du logiciel j'utilise la release disponible ici https://github.com/cedricp/ddt4all/releases
c'est la V1.5 apparemment, avec affichage en noir (sur les captures d'écrans des forums c'est en blanc)

cedricp, 2020-05-07:
Tu utilises bien le mode "connecte" au demarrage ? (pas le mode edition ?)

agn22, 2020-05-07:
oui, je clique sur le mode connecté après avoir coché le disclaimer et j'arrive sur cette fenêtre :
[[image]](url)

cedricp, 2020-05-07:
Apres, il faut selectionner le modele de vehicule et cliquer sur la loupe pour scanner les ECU.

agn22, 2020-05-07:
ok, merci, je pensais que ça scannais d'abord...quel boulet je fais, désolé.
je tente ça dès que je peux :)

agn22, 2020-05-10:
bonjour, j'ai finalement réussi à activer android, le tuto que j'avais suivi ne donnait pas les manips dans le bon ordre.
cependant le bouton expert reste grisé même s'il est possible de l'activer. peut être un bug d'affichage dû à l'interface sombre (sur les captures de forum on peut voir que le logiciel était en blanc avant) ?  je joint une capture
[[image]](url)
Merci en tout cas pour ton logiciel bien utile, je vais creuser les autres possibilités qu'il nous offre :)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif