cedricp/ddt4all #385: rLink1 video in motion

sebomaster, 2020-05-16:
In my ddt4all in video restriction i have 2 choice : restricted and limited feature. I not have permited button. Why?

Marianpol, 2020-05-16:
Maybe your ECU does not support this function? Can you tell me which file exactly was found?

mnigten, 2020-05-16:
I think this has to do with if it is permitted you would be albe to watch movies while driving. Therefore they use the limited option

sebomaster, 2020-05-16:
Navi 4.6

sebomaster, 2020-05-16:
I,m in demo mode and the same options

sebomaster, 2020-05-16:
Maybe my data base is wrong? From 2019. My car is from 2018 renault twingo gt 3

sebomaster, 2020-05-16:
[image]
Why?

sebomaster, 2020-05-16:
[image]
Why?

Marianpol, 2020-05-16:
There is simply no Permitted option in that ECU.
[image]

sebomaster, 2020-05-16:
Why?
Anybody know where is button deactivated restriction video in motion?

sebomaster, 2020-05-16:
Limited feature? What is this?

I-am-druny, 2021-11-23:
let me know if you found out. i have the same problem

AndyIreland, 2021-11-23:
You could try this - it may work ... it may not work but give it a go - let us know (do at your own risk - always chance could brick rlink) 

Choose mfd v1.8 - do you see there you have 'permitted' and 'restricted' ? - try writing 'permitted' then press the home button on the rlink 1 to reboot it , then see if it works when driving along ?

[image]

I-am-druny, 2021-11-23:
I'll give it a try tomorrow and tell you. thanks for your help!

I-am-druny, 2021-11-24:
[image]

I-am-druny, 2021-11-24:
[image]

it shows that its already permitted but it doesn't work 🤔

I-am-druny, 2021-11-24:
btw whats the deal with the rs display? 
[image]

I-am-druny, 2021-11-24:
i have a laguna 3. would that work?

AndyIreland, 2021-11-24:
> it shows that its already permitted but it doesn't work 🤔
Thats a shame . It was worth a go. 

Maybe you could give this a go
https://www.navinc.nl/9-tv-in-motion/video-in-motion-module-renault-r-link-system-can

I have never got the RS configuration working on my rlink1 for my Zoe .. I only think it works on RS turbo renault models , more than likely relies on extra module / hardware

I-am-druny, 2021-11-24:
I'll keep it as an option. next step is that i root my phone to unlock apps like kodi and vanced youtube for android auto. if that works, I'm more than happy

I-am-druny, 2021-11-24:
can ddt4all unlock rlink apps that you can normally buy in the rlink store? there was an app that lets you see the 4control work in real time but it isn't available anymore. actually I can't find any apps for my car in the rlink store

AndyIreland, 2021-11-24:
> can ddt4all unlock rlink apps that you can normally buy in the rlink store? there was an app that lets you see the 4control work in real time but it isn't available anymore. actually I can't find any apps for my car in the rlink store

Not sure about the apps for the Laguna 3 sorry , because the rlink 1 in my car is Zoe EV and there are loads  of different apps for that , but EV related

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif